prusa3d/Prusa-Firmware #1: X_MAX Y_MAX Z_MAX pins.

GilesBathgate, 2016-03-28:
In pins.h  the following pins are defined:

```c++
  #define X_MAX_PIN 24
  #define Y_MAX_PIN 23
  #define Z_MAX_PIN 30
```

This contradicts the pinout: http://reprap.org/mediawiki/images/e/ef/MiniRambo1.3a-schematic-pg2.svg

It should be:

```c++
  #define X_MAX_PIN 30
  #define Y_MAX_PIN 24
  #define Z_MAX_PIN 23
```

GilesBathgate, 2016-03-28:
Perhaps the pinouts vary depending on the variant used. In which case these should be defined appropriately in each of the variant header files.

ayourk, 2016-12-27:
It looks like this has been fixed as of the commit https://github.com/prusa3d/Prusa-Firmware/commit/3c735786e368c14dba5fe9115b3bafb730176f58 .