prusa3d/Prusa-Firmware #2: Enabling Auto Bed Leveling breaks the build

severeon, 2016-04-06:
I can compile version 2.2.2 successfully with bed auto leveling enabled.  The latest version breaks when I enable auto bed leveling with the following error:

```
Arduino: 1.6.8 (Windows 10), Board: "RAMBo"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\sever_000\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\sever_000\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\sever_000\Documents\Arduino\libraries" -fqbn=rambo:avr:rambo -ide-version=10608 -build-path "C:\Users\SEVER_~1\AppData\Local\Temp\buildb09b91c15fcaebe254ec6eb9e6435f6b.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\sever_000\Downloads\Prusa-Firmware-master\Firmware\Firmware.ino"
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware "C:\Program Files (x86)\Arduino\hardware" -hardware "C:\Users\sever_000\AppData\Local\Arduino15\packages" -tools "C:\Program Files (x86)\Arduino\tools-builder" -tools "C:\Program Files (x86)\Arduino\hardware\tools\avr" -tools "C:\Users\sever_000\AppData\Local\Arduino15\packages" -built-in-libraries "C:\Program Files (x86)\Arduino\libraries" -libraries "C:\Users\sever_000\Documents\Arduino\libraries" -fqbn=rambo:avr:rambo -ide-version=10608 -build-path "C:\Users\SEVER_~1\AppData\Local\Temp\buildb09b91c15fcaebe254ec6eb9e6435f6b.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "C:\Users\sever_000\Downloads\Prusa-Firmware-master\Firmware\Firmware.ino"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_RAMBO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Users\sever_000\AppData\Local\Arduino15\packages\rambo\hardware\avr\1.0.0\variants\rambo" "C:\Users\SEVER_~1\AppData\Local\Temp\buildb09b91c15fcaebe254ec6eb9e6435f6b.tmp\sketch\Firmware.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_RAMBO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Users\sever_000\AppData\Local\Arduino15\packages\rambo\hardware\avr\1.0.0\variants\rambo" "C:\Users\SEVER_~1\AppData\Local\Temp\buildb09b91c15fcaebe254ec6eb9e6435f6b.tmp\sketch\BlinkM.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_RAMBO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Users\sever_000\AppData\Local\Arduino15\packages\rambo\hardware\avr\1.0.0\variants\rambo" "C:\Users\SEVER_~1\AppData\Local\Temp\buildb09b91c15fcaebe254ec6eb9e6435f6b.tmp\sketch\ConfigurationStore.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_RAMBO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Users\sever_000\AppData\Local\Arduino15\packages\rambo\hardware\avr\1.0.0\variants\rambo" "C:\Users\SEVER_~1\AppData\Local\Temp\buildb09b91c15fcaebe254ec6eb9e6435f6b.tmp\sketch\Firmware.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_RAMBO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Users\sever_000\AppData\Local\Arduino15\packages\rambo\hardware\avr\1.0.0\variants\rambo" "C:\Users\SEVER_~1\AppData\Local\Temp\buildb09b91c15fcaebe254ec6eb9e6435f6b.tmp\sketch\LiquidCrystal.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_RAMBO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Users\sever_000\AppData\Local\Arduino15\packages\rambo\hardware\avr\1.0.0\variants\rambo" "C:\Users\SEVER_~1\AppData\Local\Temp\buildb09b91c15fcaebe254ec6eb9e6435f6b.tmp\sketch\MarlinSerial.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_RAMBO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Users\sever_000\AppData\Local\Arduino15\packages\rambo\hardware\avr\1.0.0\variants\rambo" "C:\Users\SEVER_~1\AppData\Local\Temp\buildb09b91c15fcaebe254ec6eb9e6435f6b.tmp\sketch\Marlin_main.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_RAMBO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Users\sever_000\AppData\Local\Arduino15\packages\rambo\hardware\avr\1.0.0\variants\rambo" "C:\Users\SEVER_~1\AppData\Local\Temp\buildb09b91c15fcaebe254ec6eb9e6435f6b.tmp\sketch\Sd2Card.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_RAMBO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Users\sever_000\AppData\Local\Arduino15\packages\rambo\hardware\avr\1.0.0\variants\rambo" "C:\Users\SEVER_~1\AppData\Local\Temp\buildb09b91c15fcaebe254ec6eb9e6435f6b.tmp\sketch\SdBaseFile.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_RAMBO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Users\sever_000\AppData\Local\Arduino15\packages\rambo\hardware\avr\1.0.0\variants\rambo" "C:\Users\SEVER_~1\AppData\Local\Temp\buildb09b91c15fcaebe254ec6eb9e6435f6b.tmp\sketch\SdFatUtil.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_RAMBO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Users\sever_000\AppData\Local\Arduino15\packages\rambo\hardware\avr\1.0.0\variants\rambo" "C:\Users\SEVER_~1\AppData\Local\Temp\buildb09b91c15fcaebe254ec6eb9e6435f6b.tmp\sketch\SdFile.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10608 -DARDUINO_AVR_RAMBO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Users\sever_000\AppData\Local\Arduino15\packages\rambo\hardware\avr\1.0.0\variants\rambo" "C:\Users\SEVER_~1\AppData\Local\Temp\buildb09b91c15fcaebe254ec6eb9e6435f6b.tmp\sketch\SdVolume.cpp" -o "nul"
[... cut ...]
Marlin.h:67: error: array must be initialized with a brace-enclosed initializer

 #define SERIAL_PROTOCOLPGM(x) (serialprintPGM(PSTR(x)))

                                               ^

C:\Users\SEVER_~1\AppData\Local\Temp\buildb09b91c15fcaebe254ec6eb9e6435f6b.tmp\sketch\Marlin_main.cpp:2101:13: note: in expansion of macro 'SERIAL_PROTOCOLPGM'

             SERIAL_PROTOCOLPGM(MSG_BED);

             ^

exit status 1
initializer fails to determine size of '__c'
```

Any assistance would be greatly appreciated

primalmotion, 2016-04-19:
+1

Same thing here

primalmotion, 2016-04-20:
I had to replace 
`SERIAL_PROTOCOLPGM(MSG_BED);`
by 
`SERIAL_PROTOCOLPGM("Bed");`

At line 2010 in `Marlin_main.cpp`

Not optimal, but I guess this is because there is no default language.

severeon, 2016-04-20:
That's how I solved the issue.  I posted this a while ago to the Prusa forums.

After a few more issues similar to this one I abandoned my goal of using this as my starting point.  Instead I found it easier to migrate the prusa tweaks over to base Marlin.

primalmotion, 2016-04-20:
What  didn't work out? I sucessfully built everything and it seems to be ok. Still waiting for the sensor to arrive, but the firmware seems to be good to go

severeon, 2016-04-20:
The build was fine once that fix was in place. 

The G29 command is a bit more robust in upstream Marlin and I wanted to play with mesh bed leveling too.

PavelSindler, 2017-01-09:
I wonder why do you need to enable the auto bed leveling? The mesh bed leveling and the auto bed leveling functionalities are mutually exclusive. What exactly do you try to achieve?

closed 2017-02-14 by PavelSindler