prusa3d/Prusa-Firmware #4: Firmware version 3.0.0 unloads filament on Stop

sdjmchattie, 2016-06-02:
When the user chooses to Stop the print in Firmware 3.0.0 on a Prusa i3 Plus, part of the stop process ejects the filament like an unload,  This is because the `G1` command designed to move the X and Y axes also sets the Extruder to zero.  This causes the extruder to wind backwards for all the filament that has been printed so far when the gcode is using absolute extruder positioning as it does in Cura generated gcode.

The issue is on line 504 of ultralcd.cpp:  https://github.com/prusa3d/Prusa-Firmware/blob/dev/Firmware/ultralcd.cpp#L504

Either the extruder needs to be instructed to operate in relative mode by enqueuing `M83` or the `E0` needs to be removed from the `G1` command.

sdjmchattie, 2016-08-06:
Is there a reason why you've closed this without comment? This fix is needed, especially for Cura users where the software uses absolute extrusion values.

michalprusa, 2016-08-10:
Sorry, I forgot to hit "Comment" button. I closed that because this issue is fixed in new version.