prusa3d/Prusa-Firmware #9: Arduino Compatibility

Eriobis, 2016-09-07:
Hi, I had a hard time compiling your FW because I dian't want to change much thing that were supposed to work.
the fpos_t structure is already in the core of arduino. I suggest to change the 5 instances of this fpos_t for filepos_t to avoir any problem with the original arduino files.

Cheers

bubnikv, 2016-09-08:
We currently compile our firmware with Arduino 1.6.8. We are aware that the firmware is not compatible with a newer Arduino installation. We will fix it, but it is currently not our priority due to our limited developer resources.

ayourk, 2016-12-27:
Check my latest commit in my pull request:
https://github.com/prusa3d/Prusa-Firmware/pull/31
I've added the fix to this as per @Eriobis's request.

ayourk, 2017-04-19:
The fix for this should now be merged into the main tree via https://github.com/prusa3d/Prusa-Firmware/commit/ab4f5d25cc4b8c98b21127d960e6eea43e3e384a

mdjurfeldt, 2017-06-28:
I just (b3eeac12) tried to compile your firmware. I used Arduino 1.5.6. I then discovered that your addon for 1.5.x contained the LiquidCrystal library. In order for the compilation to go through I both had to remove the Arduino 1.5.6 LiquidCrystal library and the one in your addon. Selecting the RAMBo board and uploading this to the printer worked.

Then I discovered that your 1.6.x addon seems to contain different things than the 1.5.x one.

Is it OK for me to use Arduino 1.5.6 with the 1.5.x addon, or should I use Arduino 1.6.8 like you do? Maybe this should also be made clear in the README.md?

Best regards

JohnOCFII, 2017-06-28:
@mdjurfeldt - Check out the following issue for more information on compiling the Prusa i3 firmware.  It's still a user-developed process, and it may not be the most efficient way to get there, but it does work.  https://github.com/prusa3d/Prusa-Firmware/issues/29#issuecomment-294370913

mdjurfeldt, 2017-06-28:
@JohnOCFII Thanks for the info!

However, note that my setup as described above works with 1.5.6. I just wondered if it makes any difference to using 1.6.8. Do you know if the Arduino addon differences between 1.5.6 and 1.6.8 matter? The only thing I could spot that could potentially make a difference was some additions to u8glib.

JohnOCFII, 2017-06-29:
@mdjurfeldt - Sorry - I don't know.

3d-gussner, 2019-01-08:
@Eriobis I think this issue is outdated, you might want to close it

closed 2019-04-11 by michalxfanta