prusa3d/Prusa-Firmware #10: Parts fan keeps blowing after stop print

Brekel, 2016-09-26:
It seems like (at least on the i3 MK2) the firmware doesn't turn off the parts fan when stop print is selected from the menu.

Of course it can be manually shut off but since printing has stopped an automatic M107 could be convenient.

bubnikv, 2016-09-26:
Thanks, we are aware of this issue.
Vojtech

Brekel, 2016-09-26:
Great work btw, by far the best firmware I've seen on 8-bit and lots of cool innovation!

bubnikv, 2016-09-26:
Thanks, we appreciate it.
Vojtech

ayourk, 2016-12-27:
Looks like this was fixed as of commit https://github.com/prusa3d/Prusa-Firmware/commit/ea197360ac851975da76fd27f7e3d2be2debb72e#diff-68a5b34f911d0dfba612ae98d305faca

mrjohnc, 2017-11-08:
Bump, suggest this is closed :) @bubnikv

closed 2018-10-31 by inquisitiveKangaroo