prusa3d/Prusa-Firmware #14: No "Error:" reported on serial line in case of thermal runaway?

foosel, 2016-10-10:
I've had some users complain [on OctoPrint's mailing list](https://groups.google.com/d/msgid/octoprint/e316ed23-bea7-4718-ab40-239a3e32506b%40googlegroups.com?utm_medium=email&utm_source=footer) that a thermal runaway on their MK2s wasn't recognized by OctoPrint, which therefore kept bombarding the printer with commands and didn't stop printing.

I have still not gotten a log from the users in questions of what went over the serial, but I did a quick look into your MK2 branch here and found this function which I perceive to be the handler for thermal runaways:

https://github.com/prusa3d/Prusa-Firmware/blob/MK2/Firmware/temperature.cpp#L1163

If that assumption is correct, I'm either missing something crucial here or a thermal runaway produces NO output at all on the serial line, leaving any potentially connected hosts completely in the dark that something is amiss. If that is true, that's an issue - hosts depend on the firmware to inform them about any issues the host cannot detect since it doesn't have the data via a message prefixed with `Error:` in order to react accordingly, without such a message it's impossible for a host to do the right thing.

r0ndL, 2016-10-11:
I believe I'm the original bug reporter/user who reported this issue.  Attached is the related terminal communication around the time period which the MK2 firmware detected a Thermal Runaway.

[thermal-runaway-no-error.txt](https://github.com/prusa3d/Prusa-Firmware/files/521999/thermal-runaway-no-error.txt)

foosel, 2017-01-11:
Actually looks more like it was solved in da5f1f71eab686c45b3d2e8e743c67e4a608499c but as I do not own a MK2 myself, I cannot test this. I'll therefore leave it up to the maintainers here to close this ticket ;)

bubnikv, 2017-01-11:
Thanks for taking the burden of finding the exact commit from our
shoulders, Nina :-)

foosel, 2017-01-11:
It's "Gina" actually, but you are welcome nonetheless ;)

closed 2018-10-31 by inquisitiveKangaroo