prusa3d/Prusa-Firmware #16: Mesh bed leveling and PINDA probe temperature drift

biasedlogic, 2016-10-18:
Detailed information about the issue and workaround is here: http://shop.prusa3d.com/forum/improvements-f14/pinda-probe-live-z-calibration-z-offset-and-temper-t2048.html

Short version: the probe has a temperature drift, if it  gets hot from the bed, the offset will change. Workaround: Keep the probe away from the hot bed before mesh leveling to avoid heating it up too much. 

Marek

mgroene, 2016-10-31:
I noticed this temperature drift on my printer, too. It's not an original prusa but I'm using an inductive probe, too. My workaround to minimise the influence is next to keep the sensor away from the heated bed to turn on the fan while starting up.
This combination of both makes my sensor measure the distance to the bed quite stable and works good for me. Thank you for reporting, thought I am the only having such problems.

PavelSindler, 2017-01-10:
Thank you. We are aware of this problem and it is one of the key things on our to do list now.

biasedlogic, 2017-01-10:
Three months to even acknowledge an issue that has been documented, reproduced and dished out with a tested solution... Wow. Thank you, i guess. 

And no, it's not an 'enhancement', it's a bugfix for a design flaw.

wschadow, 2017-04-15:
Thanks Marek, for pointing this out and to even provide a solution! I ran into this issue the more I printed ABS with a higher bed then PLA. When I forgot to move to PINDA at least 4-6 cm above the bed, I had to adjust Life-z or the print is ruined.
As far I noticed this issue is now even a three more months not addressed, although being a "key thing". That's bit strange, isn't it?

cjshaker, 2017-04-28:
It would seem that an easy way to fix this would be to change the default 'Home' position to have a higher Z value of 5 cm above the print bed or so. That would also make it easier to inspect and clean the print head before starting your next print?

Chris Shaker

wschadow, 2017-04-28:
yes, I  have an adapted start script that moves the print head a few cm above the bed, heats the bed and then does mesh bed leveling.

G28 W ; home all without mesh bed level
G0 X60 Y-3 Z80; move extruder above bed to keep PINDA cool keep extruder in front for cleaning

M104 S160 ; set extruder temp to 160C preheat nozzle
M140 S[bed1_temperature] ; set bed temp for temperature of the first layer
M109 S160 ; wait for extruder temp
M190 S[bed1_temperature] ; wait for bed temp

M104 S[extruder0_temperature] ; set extruder temp for temperature of the first layer
M109 S[extruder0_temperature] ; wait for extruder temp

G80 ; mesh bed leveling including homing all 3axis

G1 Y-3.0 F1000.0 ; go outside print area
G1 X60.0 E9.0 F1000.0 ; intro line
G1 X100.0 E12.5 F1000.0 ; intro line

biasedlogic, 2017-04-29:
Wschadow: that's pretty much copy of the code I've posted in the original workaround description. Good to know it  works for others too! For some time I had the impression I'm the only one seeing the problem...

biasedlogic, 2017-04-29:
Seems I can't edit post on mobile... I meant the solution you posted uses same pattern, so it seems the workaround is good, didn't mean to sound like I 'owned' the solution, sorry.

3d-gussner, 2017-04-29:
The firmware actual build 3.0.10-11 has a PINDA temperature tuning function and it looks very promising. It corrects the Z-height readings during mesh bed leveling with different heated bed temperatures.

wschadow, 2017-04-29:
biasedlogic sure our both ways are similar. Two weeks back I had seen your script but couldn't remember where I had seen it.  Yesterday I saw the comment by cjshaker, didn't read older posts and thought it was good idea to post a solution here. Sorry, when not referring to your solution as well.

biasedlogic, 2017-04-29:
Yeah, sorry, I have rewritten and deleted some text and in the end it sounded stupidly, like I wanted to accuse you of copying, while what I really wanted to underline was, that it seems more than one person came to almost identical solution, which means the problem is legit and the solution seems to work for more than one person.

wschadow, 2017-04-29:
no problem.

markaswift, 2017-05-07:
It makes more sense to heat the nozzle to 1st layer temp AFTER the bed levelling and before the purge line. Avoids heated filament getting deposited on the bed during the levelling process.

biasedlogic, 2017-05-08:
It depends. If you wait for the nozzle first, there's a bit more time for the bed temperature (and, hence, dimensions) to stabilise. Unless you have nozzle oozing problem, it's better to do it in this sequence.

Sebastianv650, 2017-07-03:
I did some testing today to see what repeat accuracy I can get from the probe, with and without temperature changes.
Basicaly I get the same results as already discussed here: The repeat accuracy is quite good, but only if temperature isn't changing. Even after two minutes of preheating the PINDA probe as it's done with temperature calibration enabled, the returned values still changes over time of the probe isn't moved away from the heated bed.

Therefore I will also adjust the start gcode to keep the nozzle away from the heat bed and therefore as cold as possible. But - why everybode here seems to do the MBL after bed heat up? It should result in even more steable readings when we do the MBL before setting the temperatures?
Or I'm missing something..?

udvlp, 2017-07-03:
@Sebastianv650 I think the bed deforms when heated, so it's important to measure the heated bed.

biasedlogic, 2017-07-03:
Yep. Bed deforms (warps) when heated, the material of the bed above the reference pads expands too, there's about 50μm offset in z dimension per 50 degrees Celsius in my printer

Thijxx, 2017-08-22:
So whats the story on this year old ticket, it is one of the "key things on our to do list" since 9 months @PavelSindler, and where is the "PINDA temperature tuning" that @3d-gussner mentioned?

biasedlogic, 2017-08-22:
@thijxx
I know there's an effort to get the temperature compensated, but to be honest I won't trust it at present. There's no temperature sensor at the pinda probe, so any compensation must be based off a temperature model and since it's an open frame printer no model will work reliably. Just raise the print head higher after print and you'll be fine and compensate only for the hear expansion of the bed itself

Thijxx, 2017-08-22:
ok thanks

3d-gussner, 2017-08-23:
@Thijxx You can activate the PINDA temperature tuning under the calibration menu.
I tried it for a while and as long the heated bed isn't getting 'cold' again it worked quite good, print after print (also changing between PLA and ABS). Sometimes i had to adjust the Live-Z a bit.
BUT at the end i disabled it because after turning off the printer and the last Live-Z i got scratches on the heated bed. I have to adjust Live-Z between PLA and ABS sometimes to additional 500-800um and **DON'T** forget to set the Live-Z back to PLA settings !!
Conclusion is that it works kind off but it is still not THE solution we are looking for.

filipmu, 2017-11-25:
I tried something different, but would not recommend it - it caused the nozzle to crash.  Posting it here in case it helps find another approach.

I thought I would try doing the PINDA bed leveling only at the start of a print session, before the first print, and then assume it will be good for subsequent prints.   For subseqent prints, I tried to home only X and Y, leaving Z alone. This way heating will be less of an issue.  Approach I tried:
1. Run the bed leveling from the LCD on a cold bed.  My limited experience has been that thermal expansion of the extruder changes the gap between bed and nozzle more than the bed itself.
2. Do a print with the following gcode header:
M115 U3.0.12 ; tell printer latest fw version
M83  ; extruder relative mode
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
;G28 W ; home all without mesh bed level  - note commented out
G28 XY ; added this line - only home X and Y axis, since Z assumed good
;G80 ; mesh bed leveling - note commented out - no bed leveling 
;G1 Y-3.0 F1000.0 ; go outside pritn area - commented out
G1 Y-3.0 Z 0.20 F1000; go to outside print area and move Z to print height
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line

This did not work as intended - after the G28 XY command, the next command gouged the print bed.  The display said Z was at 0.20 but it was lower than that.  I found the G28 XY affects the Z axis homing as well.  I think it modifies the mesh bed leveling results.  Seems like a defect.

biasedlogic, 2017-11-26:
>  I found the G28 XY affects the Z axis homing as well. I think it modifies the mesh bed leveling results. Seems like a defect.

Yep, the homing and leveling state machine in the firmware has more bugs than a stray dog. I have up on debugging it after it dug a second hole in my print bed.

Crunch69, 2018-04-05:
this is a major issue why is it taking so long to fix this bug that prusa is aware of for so long?

biasedlogic, 2018-04-05:
They kind of did fix it: you can now buy an upgrade to a probe with a temp sensor...
Just far as I know nobody ever acknowledged this as design flaw

Crunch69, 2018-04-06:
where is the probe temp sensor biasedlogic is speaking of?

biasedlogic, 2018-04-06:
@Crunch69 the sensor is integrated in the new PINDA probe that ships with Prusa i3 MK3 and with the upgrade kit MK2.5

closed 2018-12-11 by inquisitiveKangaroo