prusa3d/Prusa-Firmware #18: Unable to set PID settings using M301

fiveangle, 2016-11-16:
Using 3.0.9 GA on current gen MK2

M301 submission to update P receives gibberish in response followed by echo back that the new value was received, but M504 shows values unchanged.  Using M501/M500 shows same unchanged values:

```
Send: M301 P35
Recv: )}f7}_]
z^{Z]v=^sG$^kxrx_{/lk-9W;V6f{_fnq_!5?wO;e,NetO>7wj8>u|Z/^~i>Nz
~}w+{N7|{?z]Uku_;G{rT2s*?~n?zI])v/-k(tl(r]g\ytJh.mw?{
U:i7/OHlmwE9mG~`bwg$/:)7=gcIuk}?X{4_MV8M[li(GtWL}sN_<\;s_gi_Zw^#W|u|D9p17q}n9BQvV_wVgc~vg+-Yv]{-weq{9vl7k}w9?#}}>/}z~u<?qyf_m*UO4o{}%o`o<T0}?wY;=%~j^g3e|OBz.(]?p=;Wex?Vitm~q-t};w:6ppf}1c6xoK}L}DUln}p?>Ta\8QSwfUw
Recv: ;_r8{h-w )3}5Wo[gg~_y_cOg3_e6d~l81W5|~qy=u7?^w/mW'{/RiYaO^iVOVb/]5Q_zi^p[?53M..?9xg:K4/W/#nYwZ'w^1Y'u}?~>~}hylo}-7{9Qy)f{Ts?>7e.Uquv0yO9r##87Dc?;rx}no'[{'i.FzkW=]~_GzGy}N__ue/y_[mvfv`/o={o_h|i}..fo_wll/r}VK7X=/Og&/+jM_&p}.g/u:ua)u7s};9[[{}xnpKy6Zo\~3y.Xk{f?&{=lkO$j_j/>=7?2W^{+s*^_Z}Vs/SM}7!y[_@[o  p:35.00 i:4.88 d:86.08 c:1.00
Recv: ok
```

nicw, 2016-12-02:
I have this issue as well. 

`Send: M301 P35
Recv: =g�f����/���w\���.���w^_?^�>N���}#���6.��Z�����6~=[��v���9=���������_g�����g����/�G�vn��k�Y�k/�������8�����x�+>9��}�Z��rW]����<qzQ�jN|�^���G���~�S[��e��Y��J��\�������l����J�������<���{��s�i������^��MhX�ysV��=o���m������B�������������}&c����-O�p����=��yM�����d���k;`�L�ogv�K��xM�F�����^��qfgV�v��f=��8�(����i������q5v���"��_�v����g�5@�����.��?)�i@+��U���I����.�m�}S�k�}�����]n����W7�k���3��{kg?����7�3\�9�����v]������������}�i�+[&��u�WC?��h��{���~��K;m9���o>���^���L��N��Z��mw}�_���:��^������}�����?}����F���c�qHq��cOE����z���J���7���;��]_�$������Se��W�j���o�����q����Qo�T��n�������k�a�2y#|��9~R'�������?��^���=r{����;0��?WJ�����Nn|pn�sb�_0�{H�~����w�����u����L�Y�^���u�m1+������W��w����?����R�w?���s{�n��t��G����d�����~?����;��������ZO�}��(����B;p�k}�Xw�Y�ZY�-`7��gw��o��/�/���[���,�G����,S~i���&^yg��?Ou��}�������e�mUj>o|�d�����o���e�%��-�y'�.�]de���*q�lc�����u]�|aL�&�o��(_�����6���=���?��=���7k��~���7��cR!�|���fu�>/P��X������F��o�����Jr���7{������;����XH���m\������mc=����p����|_���_�6�w�y]_�g�wN�����{�{�����v��V���O'�o�J�r����q��'�pLj�u��8��������w�������S���L����w�x������&���k�b���o�MS�qKZV�t������mO��H�?G�����b��������������;������7��^����|�k��:�����l�_Q��t���;(hY	���n�Hy
Recv: ?O�~�������K��O{(a��=^��\���m��s>|�X�����9�j��7�[������8U������\�};6�f�_�[�2��zQ�����=����6?�����n�~7����\�J7�?�������G�����dw
����}�z������������q��}�o������c������~���%�][�o��I����^_����l��<�Wy����#�_�~��������l������8/����{����������=����������;��lqvc^���zLC��<_X3���W�_�a]��\���[�dw� p:35.00 i:4.88 d:86.08 c:1.00
Recv: ok`

fiveangle, 2016-12-02:
Interesting how it's a completely different response.

fiveangle, 2016-12-02:
Are you running 3.0.9 @nicw ?

nicw, 2016-12-02:
@fiveangle yes, 3.0.9

fiveangle, 2016-12-12:
It appears that even if they address this issue, we will not be able to save the corrected settings since EEPROM is disabled (see #21), so we will have to use slicer Start GCODE script to accomplish.

It seems the correct path Prusa Research expects is for us to submit a support ticket to have them start replacing hardware if the hotend or heated bed cannot stay regulated.

bcheck555, 2016-12-14:
I am seeing this problem as well. I have gone back to the older firmware and am able to put my PID settings into Slic3r to set before the print starts. My extruder temps have gone from +-4*C to +-0.5*C.

fiveangle, 2016-12-14:
Which FW did you downgrade to @bcheck555 ?  3.0.8 ?

bcheck555, 2016-12-14:
@fiveangle Correct I am running v3.0.8 currently.

ryanneufeld, 2016-12-22:
Problem occurs in 3.0.9 as well

PavelSindler, 2017-01-10:
Thank you very much for reporting this. M301 and M304 has been fixed in 3.0.10. Now it is also possible to store PID settings using M500.

closed 2017-01-10 by PavelSindler

chixxi, 2017-01-30:
Did I understand this correctly: Has been fixed in 3.0.10_alpha?

If yes, when can we expect the update?

geekmoto13, 2017-02-20:
M304 is not saving with the m500 this is still a bug.  m301 does save however

terminal output from octoprint
`
[...]
Send: M301
Recv: ok p:18.11 i:1.38 d:59.35 c:1.00
Recv: ok
[...]
Send: M304
Recv: ok p:126.13 i:4.30 d:924.76
Recv: ok
m301 is the current tuning for the hotend
m304 is the firmware values

prior  to reboot
Send: M304 P59.44 I2.43 D362.74
Recv: ok p:59.44 i:2.43 d:362.74
Recv: ok
[...]
Send: M500
Recv: echo:Settings Stored
Recv: ok
[...]

after reboot

Send: M301
Recv: ok p:18.11 i:1.38 d:59.35 c:1.00
Recv: ok
[...]
Send: M304
Recv: ok p:126.13 i:4.30 d:924.76
Recv: ok`