prusa3d/Prusa-Firmware #21: M500 and EEPROM functions

pboschi, 2016-12-08:
M500 function can't be enabled; the max_*_jerk variables don't exist (they have to be renamed) and also some variable declarations about preheat temperature are kept off by some #define.
Anyway, after fixing those bugs, the M500 still doesn't store values into eeprom (e.g., PID tuning values).

bubnikv, 2016-12-08:
The Prusa3D firmware does not store any usual Marlin configuration values
into EEPROM as of today. This is intentional, the configuration values are
expected to be set to the optimal values by the manufacturer.

nicw, 2016-12-12:
Hi, I'm coming over here from [Unable to set PID settings](https://github.com/prusa3d/Prusa-Firmware/issues/18), which are settings that are very specific to an individuals printer. We need to be able to set those in EEPROM.

I understand that Prusa wants to have an optimal configuration, but **this is not a product meant to work out-of-the-box; Prusa _literally_ uses zipties to hold its product together**. Tolerance is too great to consider anything optimized. 

Please reconsider blocking users from setting EEPROM values.

ryanneufeld, 2016-12-22:
@bubnikv I can tell you that the manufacturer PID values are 100% not optimal for every printer.

With the stock values I was seeing 2-5 degree temp swings. Additionally now that Prusa has started quietly shipping w/ 40w heat cartridges the PID values will again not be optimal.

Please re-enable this feature, or provide instructions on how to properly build this firmware so we can do it our selves.

bubnikv, 2016-12-22:
We plan to enable the storing / loading of the EEPROM values, but there multiple things on our plate.

GreatGrizzly, 2016-12-29:
I am probably over simplifying this, but would fixing this just mean uncommenting 
#define EEPROM_SETTINGS
in configuration.h?

pboschi, 2016-12-29:
This thread is about the problems you encounter AFTER you do that!

jspiro, 2017-02-04:
Looks like this was addressed in the new release candidate!

jspiro, 2017-02-04:
@PavelSindler Thanks for addressing this btw, with my e3d socks on it's great having the tuning saved. Will this persist across firmware updates?

bubnikv, 2017-02-04:
Yes, these settings will not be influenced by the firmware update.

closed 2017-03-23 by PavelSindler