prusa3d/Prusa-Firmware #22: Optimize some jerk code

thinkyhead, 2016-12-09:
I found that some of the jerk code is redundant and can be optimized pretty easily to save some cycles in the planner. The [code in this loop](https://github.com/prusa3d/Prusa-Firmware/blob/MK2/Firmware/planner.cpp#L1003-L1022) can be changed from the existing…
```cpp
for (uint8_t axis = 0; axis < 4; ++ axis) {
  float jerk = fabs(current_speed[axis]);
  if (jerk > max_jerk[axis]) {
    // The actual jerk is lower, if it has been limited by the XY jerk.
    if (limited) {
      // Spare one division by a following gymnastics:
      // Instead of jerk *= safe_speed / block->nominal_speed,
      // multiply max_jerk[axis] by the divisor.
      jerk *= safe_speed;
      float mjerk = max_jerk[axis] * block->nominal_speed;
      if (jerk > mjerk) {
          safe_speed *= mjerk / jerk;
          limited = true;
      }
    } else {
      safe_speed = max_jerk[axis];
      limited = true;
    }
  }
}
```
to the equivalent, but more concise:
```cpp
for (uint8_t axis = 0; axis < 4; ++ axis) {
  const float jerk = fabs(current_speed[axis]);
  if (jerk > max_jerk[axis]) {
    if (!limited) {
      safe_speed = max_jerk[axis];
      limited = true;
    }
    else {
      const float mjerk = max_jerk[axis] * block->nominal_speed;
      if (jerk * safe_speed > mjerk) safe_speed = mjerk / jerk;
    }
  }
}
```

ryanneufeld, 2016-12-22:
You may want to keep the comments in there, and open a pull request?

bubnikv, 2016-12-22:
This is a sufficient form to us.
Thanks, Scott.

thinkyhead, 2016-12-24:
@bubnikv Cool. For anything more complex in future, I will make a PR for you.

closed 2018-12-11 by inquisitiveKangaroo