prusa3d/Prusa-Firmware #23: Add option to pause print away from the top layer of the print

mrjohnc, 2016-12-09:
When pausing a print the printer does not raise the hot end meaning that the hot end makes a little melted section in the print, this is made worse if the printer is paused for a long time when cooling down the hot end and the heating it back up is needed.

Is it possible to either have an option to pause the print and move the print head away from the print or to have this as the default?

mrjohnc, 2017-01-01:
Perhaps this option could be called 'long pause' or 'be quiet' or something? I can see it being very useful to people who need the printer to be quiet for a while during a long print.

pboschi, 2017-01-04:
please, add this feature!

werewolf1479, 2017-02-14:
Just adding to the want for this feature!  I have octoprint for monitoring when I'm not home, but last time I went to sleep and had a print going the printer destroyed itself.  Would be nice to be able to pause, go to the corner, let it cool down and then resume in the morning.

joeynuggetz, 2017-02-26:
I vote for this too.

pfa230, 2017-02-28:
Just got my print ruined due to this issue, so would very much like it fixed

laird, 2017-03-12:
For reference the Sailfish software (used in Makerbot Replicator & most 'clones' of that design) does this by default, moving the extruder to X/Y 0 and Max Z, and cooling, and on resume heating up, the restoring print X/Y/Z and resuming printing. Very convenient, not only for avoiding melting a spot on the print, but also for swapping filament (e.g. when you're near the end of a spool you can pause, unload, then load a new spool of filament and resume).

inquisitiveKangaroo, 2018-12-11:
The feature has not been implemented in newer firmware version. Thank you all for your feedbacks.

closed 2018-12-11 by inquisitiveKangaroo