prusa3d/Prusa-Firmware #24: When pausing a print and cooling down the hotend the cooling fan does not restart when restarting the print

mrjohnc, 2016-12-09:
When pausing a print and cooling down the hotend the cooling fan does not restart when restarting the print. It is possible to restarting it by going into the tune menu but this mean the fan runs constantly not only when needed.

ayourk, 2016-12-27:
If you are pausing an SD card print, I think I've committed a potential fix via https://github.com/prusa3d/Prusa-Firmware/pull/31/commits/6129c9167d342cf09413a04b837c58c76d10a9e5 which is in my pull request #31

mrjohnc, 2016-12-28:
@ayourk yes, this issue was from an SD card print, hope your pull request fixes the problem :)

ayourk, 2017-04-19:
The fix for this should now be merged into the main tree via https://github.com/prusa3d/Prusa-Firmware/commit/ab4f5d25cc4b8c98b21127d960e6eea43e3e384a

closed 2018-12-11 by inquisitiveKangaroo