prusa3d/Prusa-Firmware #29: Please update the build instructions

pboschi, 2016-12-20:
The build instructions need to be more precise.
For example:
1) you need a specific Arduino IDE version (1.6.8) or to do  some changes 
2) the stk500boot_v2_mega2560.hex file is missing
3) the generate_version_header_for_marlin file is missing
4) a bit of explanation on why are you doing each step usually helps

ryanneufeld, 2016-12-22:
👍

pboschi, 2016-12-23:
The following are more reasonable instructions.

# Original Prusa i3 MK2 Firmware

## General instructions

Pre-compiled hex files for all printers by PRUSA RESEARCH are available in hex_files folder.

Just clone the repo and flash it to the firmware.

## Build instructions

### Step 1 - Download and Install/Unpack Arduino

Download and install Arduino; suggested and verified version is the 1.6.8 from:
https://www.arduino.cc/en/Main/OldSoftwareReleases#previous

If you also use Arduino for other purposes, it is suggested to use a stand-alone version rather than a system-wide install (e.g., use the "Windows" version instead of "Windows Installer"), and use it only for Prusa i3 MK2 firmware generation.

### Step 2 - Download PRUSA i3 MK2 Firmware

Clone the prusa3d/Prusa-Firmware (Branch: MK2) repository to a local path.

If you are not an expert git user, it is suggested to Download it as ZIP from:
https://github.com/prusa3d/Prusa-Firmware/tree/MK2

### Step 3 - Prepare the Arduino toolchain

#### Step 3a - Remove the LiquidCrystal library

Due to name collision, you have to remove the Liquid Crystal library of the original Arduino installation. To do so, you may rename the "Arduino\libraries\LiquidCrystal" directory to "Arduino\libraries\LiquidCrystal_", or remove it altogether.

#### Step 3b - Add RAMBo hardware / library support

Copy the contents of the ArduinoAddons/Arduino_1.6.x directory from the downloaded firmware repository to the root of the installed arduino path.

### Step 3c - Add the bootloader

Download the bootloader image from:
https://raw.githubusercontent.com/arduino/Arduino/master/hardware/arduino/avr/bootloaders/stk500v2/stk500boot_v2_mega2560.hex

Save a copy of it on each of the following directories:
* Arduino/hardware/arduino/avr/bootloaders/stk500v2/
* Arduino/hardware/marlin/avr/bootloaders/

Double-check that the file has the .hex extension and not .txt.

### Step 4 - Prepare the Arduino Firmware

Copy and rename the file Prusa-Firmware/Firmware/variants/1_75mm_MK2-RAMBo13a-E3Dv6full.h file to Firmware/Configuration_prusa.h (note that you may have to use the 1_75mm_MK2-RAMBo10a-E3Dv6full.h if you have an older RAMBo board.

### Step 5 - Load and configure the Arduino IDE

Load the Arduino IDE and select the RAMBo as target board from the Tools submenu. You are now ready to modify and build the firmware!

### Step 6 [Optional] - Edit the Firmware

Make some modifications to the firmware.

For example, edit the Configuration_prusa.h file and edit (at your own risk):
* Extruder PID tuning: DEFAULT_Kp, DEFAULT_Ki and DEFAULT_Kd values
* Bed PID tuning: DEFAULT_bedKp, DEFAULT_bedKi, DEFAULT_bedKd values
* Extruder fan noise: reduce EXTRUDER_AUTO_FAN_SPEED from 255 to 96 (experiment a bit)

### Step 7 - Build the Firmware

You can now build the firmware:
* Compile the firmware with Sketch -> Build (CTRL-R).
* Export the compiled binary with Sketch -> Export compiled sketch (CTRL-ALT-S).

You will find the exported binary in the Prusa-Firmware/Firmware directory:
* Firmware.ino.with_bootloader.rambo.hex

#### Step 7a - Known errors
If you encounter the "using typedef-name 'fpos_t' after 'struct'" problem, you either have a wrong Arduino version (need 1.6.8 - check Step 1) or need to manually modify fpos_t to fpos_t2 in the files SDBasefile.cpp and SDBasefile.h.

If you are using Linux AND using a cloned git repo, you'll encounter the "generate_version_header_for_marlin execute file not found in $path" error; download and install the script from the following website:
https://github.com/Wackerbarth/Marlin/issues/1
You'll also have to make the script executable (chmod a+x path/to/file)

### Step 8 - Upload the Firmware

Upload the new firmare to the printer using the standard firmware update tool included in original Prusa firmware:
http://manual.prusa3d.com/Guide/Upgrading+firmware/66#_ga=1.48414886.1605935719.1473440900

bubnikv, 2016-12-24:
Thanks a lot, @pboschi

ryanneufeld, 2016-12-24:
@bubnikv you can confirm then that these steps are accurate?

bubnikv, 2016-12-24:
Not now, but we will let someone on the team do that after the Christmas. We have new hires in the software team, who could now share our burden.

ryanneufeld, 2016-12-24:
That's great news!

boelle, 2017-01-01:
i want to give this a go on my mendel90

i'm no expert but what is it that makes the XYZ corrections thick? is it a special bed? and can i get that on my own?

pcbroch, 2017-01-06:
I have followed the above steps on Windows 10 and got everything to compile and upload to the Rambo without issues.  Thanks @pboschi

aderusha, 2017-01-07:
@pboschi you saved my day, your guide was just the ticket to getting this working.

Using your guide, I've gone and [scripted the installation process for Windows users](https://github.com/aderusha/Get-PrusaFwDevEnv).  The script will download all the required components, modify settings, and move files to where they need to be in order to create a complete, portable Arduino IDE ready for use w/ the Prusa FW.

montreuilg, 2017-02-01:
@pboschi Thanks a lot for your guide. I'm using Arduino 1.6.8 on a Macintosh (Sierra) and when I try to compile the firmware, it stops with this error: "the generate_version_header_for_marlin file is missing: no such file or directory". I googled the error, but did not find any relevant info so I thought I might ask for help: any Mac users/devs successfully compiling the current firmware ?

jakester488, 2017-02-02:
the stk500boot* bootloader file seems to be included in Arduino-1.8.app/Contents/Java/hardware/arduino/avr/bootloaders/stk500v2/stk500boot_v2_mega2560.hex

bcheck555, 2017-03-10:
This works under Linux (Fedora 25). Thanks @pboschi !

dannycoates, 2017-03-18:
@ryanneufeld for OSX follow @pboschi's guide above but the base location you copy to will be in your Arduino app. Mine is `/Applications/Arduino.app/Contents/Java`.

 I also had to delete the file `/Applications/Arduino.app/Contents/Java/hardware/marlin/avr/platform.local.txt` for it to compile.

ss89, 2017-03-25:
You should also be fine compiling it when using version 1.6.11 of the arduino avr toolchain in the board manager. just tried it and i got some warnings, but generally it compiles fine.

filipgoc, 2017-03-29:
Thanks everyone. This thread is officially one of the more helpful parts of the internet now.

eyal0, 2017-03-29:
This needs to a PR.

phantomkingx, 2017-04-16:
This guide is absolutely perfect, flawless, impeccable.  Why not include it?

eyal0, 2017-04-16:
I took the instructions as written above and opened a PR.

phantomkingx, 2017-04-16:
Awesome!  I'm sure many possible firmware improvements and mods from the public were lost simply because they gave up on compile this.

probonopd, 2017-07-02:
Please see https://github.com/prusa3d/Prusa-Firmware/pull/136 which, if merged, will build the firmware on [Travis CI ](https://travis-ci.org/)automatically with each `git push`.

JohnOCFII, 2017-07-02:
Is Prusa Research still building with 1.6.8 and these manual steps?  I've been working (a little) towards getting it to build on 1.6.13 with as few manual steps as possible, and it compiles, but mainly I still have lots of pins.h warnings to resolve.

GurliGebis, 2017-07-02:
The description above works fine, but having it merged is unlikely, since it seems to have been ignored for the past 7-8 months by Prusa Research :(

daturach, 2017-07-04:
@pboschi :1st_place_medal:  works as described on Win 7. Thanks a lot.

AYapejian, 2017-07-26:
Have been tried to get this compiled and uploaded a couple times over the past 2 weeks or so.  I've followed the instructions from scratch each time, and everything compiles without error however when trying to upload the hex file via the firmware uploader I get the following error.   Env details below.  Also I don't think it's a USB issue as many search results state, I've tried different cables, restarting the printer (power on/off, unplug, etc..) and am able to update the official firmware every time without issue.  Any thoughts?

Arduino: 1.6.8
OS: Mac OS 10.12.5

Upload Error: (Can't paste from firmware updater so just the excerpt at the end

```
100% 28.90s PRUSAavrdude: verifying ... PRUSAavrdude: verification error, first mismatch at byte 0x3e1b9 0x79 != 0x72
PRUSAavrdude: verification error; content mismatch
PRUSAavrdude done. Thank you.
```

Panayiotis-git, 2017-10-16:
Thank you. This guide should replace the current Prusa-Firmware guide at the git

radzimir, 2017-11-03:
I have just managed to build the firmware with Eclipse ( Sloeber, http://eclipse.baeyens.it/ ). 
Setup script is commited there:  https://github.com/radzimir/PrusaEclipse
Please start with -h for help.

Pc-6, 2017-11-16:
I have followed all the directions by pboschi. The reflash was successful. In the firmware the settings were all unaltered. When attempting to set the extruder temp on the printer it would not let me exceed 295 degrees. If I'm understanding the firmware properly I should get at least 305. I'm not sure what I am doing wrong, perhaps I have been looking in the wrong place (Configuration_Prusa.hex) Ultimately I would like to be able to run up to 350 degrees and use a PT100 sensor(not installed yet, until I get a grasp on all of this). Also I need to know how to activate the proper pins to allow the board to read the new sensor.
I am running the Original Prusa MK2s and after uploading the firmware the printer shows version3.1.0
Thanks for any help.

3d-gussner, 2017-11-16:
@pc-6 The PT100 is merged into Prusa firmware checkout https://github.com/prusa3d/Prusa-Firmware/pull/31.
As soon you have the PT100 installed and configured in the Configuration_prusa.h the HEATER_0_MAXTEMP rises from 305 to 410
Read the Prusa Forum topic https://shop.prusa3d.com/forum/improvements-f14/mini-rambo-1-3-and-e3d-with-pt100-t700.html 
https://shop.prusa3d.com/forum/improvements-f14/mini-rambo-1-3-and-e3d-with-pt100-t700-s50.html#p22695

Hope that helps. Guess it would be better if you had open a new issue, because this is a different topic

Also helpfull
https://github.com/3d-gussner/Prusa-Firmware/blob/MK2-Private_build/BUILDING.md
https://github.com/prusa3d/Prusa-Firmware/pull/272

Pc-6, 2017-11-16:
I've been at it for another 7 hrs and seem to be chasing my tail. I have installed the PT100 sensor and uncommented the line for PT100 with amplifier. Now when I update the firmware my screen shows all four lines are blanketed with squares. When I cycle the power on the printer the LCD then shows the 1st and third rows with the squares. Where do I go from here?
Thanks for helping a newbie.

tgunr, 2018-01-04:
On OSX I modified the steps slightly and have it compiling with Ardunio IDE 1.85. 
My changes are at: http://confluence.polymicro.net/confluence/display/3DP/Compiling+Firmware+for+Prusa+MK2S

closed 2019-01-29 by inquisitiveKangaroo