prusa3d/Prusa-Firmware #40: M303 PID Tuning communication lost

Gobol, 2017-02-05:
3.0.10-RC1: After submitting M303 E0 S200 C8 (Octoprint terminal) I get normal PID tuning procedure response, but in cooling phase LCD status line shows M105 and USB communication drops - after MCU restart everything is back to normal.

jspiro, 2017-02-05:
I experienced the same. I restarted and gave it the M301 from the test and then M500 and all was well.

ChrisP-Git, 2017-02-06:
Using OctoPrint, you have to add M301, M303 and M304 to the long command list, in advanced settings of communication. However this wasn't enough for me, I suspect because OctoPrint send M105 command on regular basis to get temperature information, and maybe there is a case where this command interact with the PID tuning procedure. You can try using a putty connection instead from your PC, there won't be M105 command issued, I need to try that as well.

Gobol, 2017-02-06:
Well, only M303 is long running command here.

ayourk, 2017-02-13:
I prefer to do my PID tuning via Pronterface.  Less chance of timeouts happening.

jspiro, 2017-02-13:
I did my tuning in pronterface. That's where I'm.experiencing the time out.

PavelSindler, 2017-03-23:
Thank you for reporting this issue. It was fixed in https://github.com/prusa3d/Prusa-Firmware/commit/12505d0f4ed588b66571ecac94b17c4e2f35a6f2.

closed 2017-03-23 by PavelSindler