prusa3d/Prusa-Firmware #41: Add serial speed option

pboschi, 2017-02-05:
Could you please add serial speed option, in order to choose at least between 115200 and 250000?

Ludio, 2017-05-22:
I would image the max serial speed it limited by the hardware. Eg. Arduino's supported baud rates are 300, 600, 1200, 2400, 4800, 9600, 14400, 19200, 28800, 31250, 38400, 57600, and 115200. When using Octoprint 115200 is used so this seems to be the maximum.

closed 2018-12-11 by inquisitiveKangaroo

arekm, 2022-08-31:
@inquisitiveKangaroo Why closed?

(That pull request had noting to do with this issue)

Does anyone know if 250000 (or higher?) speed works well and reliably with prusa mk3 (and newer) hardware boards?

Higher speeds would be even more interesting than meatpack https://github.com/prusa3d/Prusa-Firmware/pull/2955

Prusa-Support, 2022-09-20:
Yeah, according to https://github.com/prusa3d/Prusa-Firmware/pull/184 it wouldn't really work well.

Michele Moramarco
Prusa Research