prusa3d/Prusa-Firmware #43: Have "Unload filament" reshape plug at end of filament before unloading

ceryen, 2017-02-07:
When unloading filament, there is sometimes a plug at the end of the filament that prevents the filament from being pulled through the hole at the top of the extruder body.  Would it be possible to have the "Unload filament" function briefly push the filament through the hotend to reshape the plug at the end of the filament before unloading the filament?  This would also need to be added to the M600 filament change function.

joeynuggetz, 2017-02-26:
Do you end your print jobs with an extraction? Only issue I've ever had with unloading is the melted strand that remains when you pull it out which loves to tangle itself in the fan.

ceryen, 2017-02-26:
No, I rarely end print jobs with extraction.  I typically get plugs when I've had the printer off for some time, then I preheat specifically to unload the filament.

bubnikv, 2017-02-26:
I am sure it does not quite answer the original issue, but I am usually
pulling the filament by hand using the atomic method:
http://support.3dverkstan.se/article/10-the

It has the benefit of cleaning the nozzle at each cold pull.

sylphio, 2018-05-03:
My 10 cents. On my mk3, i got a blob 100% of time (unchanged prusa setting PLA/PTEG) and it jamat top extruder hole. The only way to get a proper unload on my printer is to extrude 10mm before unloading.  So, in my case, this can be a game-changer in term of reliability.

inquisitiveKangaroo, 2018-12-11:
Extraction speeds were tuned in newer firmware versions. Also the ramming feature was adden into the slicer PE for MM prints.

closed 2018-12-11 by inquisitiveKangaroo