prusa3d/Prusa-Firmware #46: Front left calibration point not reachable

MartinMajewski, 2017-02-17:
After upgrading to the newest Firmware 3.0.10 I am unable to perform the XYZ calibration anymore.

The printer hasn't changed, no new modifications added and with all other mods and the previews firmware version it was not a problem to calibrate the printer.
Since the update of currently only one of my printers, I cannot calibrate it anymore. It goes over all the points, and as soon as one think that the calibration has finished successfully, the extruder moves to the parking position, and I get the message, that the front left point couldn't be reached. Front left means the side of the X axis stepper motor, right? How comes, that Mesh Bed Leveling and Z calibration work but not the XYZ calibration if all of them uses this front left point?

Is it safe to downgrade to the previous firmware version to test if it is firmware or hardware related?

Bye
Martin

ayourk, 2017-02-20:
I believe commit https://github.com/prusa3d/Prusa-Firmware/commit/e7bc6a0645675bf05950aed2c73a9bf1bb869a1a is causing this issue.  If you are willing, rebuild the firmware from this repo and revert the Y_MIN_POS change to your local copy and reflash to see if it fixes your issue.

fiveangle, 2017-02-20:
That error is due to inductive probe not being able move far enough past the round bed probe position without triggering the y-axis end stop.  It may be as tight as just losening the microswitch and putting rearward pressure on it while tightening the mounting bolts then trying again back before the home position change from 0.2 to 0.15 – it's really sensitive.  Regardless, you will likely have to re-adjust your threaded rods to put more distance between rear feet and the frame (the fabled "100mm" distance, which I think really needs to be 101mm in the assembly guide to ensure enough breathing room to allow calibration to complete confidently).

MartinMajewski, 2017-02-20:
Thank you for the answers and suggestions. It also appeared to me, that I will probably have to give the bed more movement range to the back by pulling the construction more backward. However, this is an unpleasant procedure, not only due to the fact that squaring the frame is a pain, but also because the PSU is mounted to the rod-bracket which in turn is aligned with the rear right cornerstone. This means that I put stress on the rods causing them to twist and skew as well as bending the Z frame. This is really a flow in the design of the MK2 which should be addressed with updated printable parts. I heard a lot of people I printed the Y frame square helper tool for, that they have trouble with calibration and it was the suggested procedure that helped.

However, I think it is still a bug/feature request for the firmware to stop the entire calibration immediately if this problem occurs because the calibration process takes a lot of time which is wasted if you have to wait until the very end to get the error message.
As the mentioned printer is currently a Geni pig for some experiments, I will try another solution next week: I will redesign the bed bracket which holds the bed to the drive belt to have a somehow adjustable end stop pusher, as I technically have at least 2 mm more space for the bed to travel.

ayourk, 2017-02-20:
I know that for me, I find it easier to change the software than to try and adjust the hardware.  I'm dead on at 100mm with my zFrame to rear y-corners.  If I put Y_MIN_POS at 2.2 as per commit https://github.com/prusa3d/Prusa-Firmware/commit/e7bc6a0645675bf05950aed2c73a9bf1bb869a1a my front points are not reachable.  If I put Y_MIN_POS at 4 which is prior than the commit, then my front points are reachable.  As for the PSU mount on the Y-axis, I keep my screws loose to mitigate skew.

MartinMajewski, 2017-02-20:
@bubnikv Thank you for the technical clarification. Maybe you could include the info "Perpendicularity is not guaranteed, but the printer can be used anyway" on the LCD screen. If I see a message "point not reachable" and I see real points on the bed and the extruder jumping from one to another my intuitive implication is: do not use the printer, or you risk a damage!

Not every user searches this issue thread on GitHub to get the clearance for operation.

bubnikv, 2017-02-20:
@MartinMajewski You are right, the message is not very explanatory. I am a software engineer, not a psychologist, I am not trained to read the mind of our customers.

misan, 2017-02-26:
I have seen this happening on a machine because y-motor was skewed due to belt tension. I fixed it with a 5mm piece of plywood in between threaded rods and motor case.

ayourk, 2017-02-26:
That's what the [y-distance.stl](https://github.com/prusa3d/Original-Prusa-i3/blob/MK2/Printed-Parts/stl/Y-distance.stl) file is there to fix.  You can even print it out of Flexible filament for noise dampening.

MartinMajewski, 2017-02-26:
I have fixed this issue already, by heating up the screws of the Y end stop and moving it backward. It was that one millimeter that was missing to get the calibration finishing without a complaint.

closed 2017-03-17 by PavelSindler