prusa3d/Prusa-Firmware #47: Unload filament function, unloads to slow

chriszero, 2017-02-18:
Since the 3.0.10, the "unload filament" is "slower", so slow that the filament is stringing on unload. I've got a clog in the cold zone for two time now.
Why is this slowed down in this release?

bubnikv, 2017-02-18:
@chriszero Why is this slowed down in this release?

Because many customers complained, that it so fast, that it causes grinding of some filaments.

I guess one size does not fit all.

chriszero, 2017-02-18:
Hmm, ok...
I will go back to the previous release.
Never got those grinding filament on unload, and if it causes some grinding it's a lesser problem than disassemble the whole extruder to clean :-(

fiveangle, 2017-02-18:
Agreed: this new routine is a disaster. If someone complained of "grinding filament" then either they had not preheated, or they had no retraction perfonrmed in their end gcode from previous print so filament has bulb on the end, and they will never get it out without extruding filament past the bulb then unloading or wait until the bulb is heat soaked. 

This must revert to 3.0.9 unload speed.

chriszero, 2017-02-18:
https://goo.gl/photos/BoJ2DcEBYjvT4hJD6
Clog no. 3....

cyberreefguru, 2017-02-20:
I've found the unload way too fast in previous releases.  Never enough time to get the filament out of the hot end without a lot of pulling.  I haven't tried 3.0.10 yet, but slower is welcomed for me.

PavelSindler, 2017-02-20:
Can you check maximum feedrates by sanding M503?
Did you use precompiled hex file or did you compile yourself? What E feedrate do you have in Configuration_prusa.h? 
Is it the same if you unload filament right after resetting the printer?

cyberreefguru, 2017-02-20:
@PavelSindler I'm not sure who you are responding to, but here are my settings with the 3.0.9 stock firmware loaded from the hex file:

Send: M503
Recv: echo:Steps per unit:
Recv: echo:  M92 X100.00 Y100.00 Z400.00 E161.30
Recv: echo:Maximum feedrates (mm/s):
Recv: echo:  M203 X500.00 Y500.00 Z12.00 E120.00
Recv: echo:Maximum Acceleration (mm/s2):
Recv: echo:  M201 X9000 Y9000 Z500 E10000
Recv: echo:Acceleration: S=acceleration, T=retract acceleration
Recv: echo:  M204 S1000.00 T1500.00
Recv: echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
Recv: echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.20 E2.50
Recv: echo:Home offset (mm):
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:PID settings:
Recv: echo:   M301 P40.93 I4.88 D86.08
Recv: echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)
Recv: echo:   M207 S3.00 F2700.00 Z0.00
Recv: echo:Recover: S=Extra length (mm) F:Speed (mm/m)
Recv: echo:   M208 S0.00 F480.00
Recv: echo:Auto-Retract: S=0 to disable, 1 to interpret extrude-only moves as retracts or recoveries
Recv: echo:   M209 S0
Recv: echo:Filament settings: Disabled
Recv: ok

-Tom

PavelSindler, 2017-02-23:
Thank you Tom. It seems ok.
Can you please provide additional information @chriszero and @fiveangle?

chriszero, 2017-02-23:
From 3.0.10
SENDING:M503
echo:Steps per unit:
echo:  M92 X100.00 Y100.00 Z400.00 E161.30
echo:Maximum feedrates (mm/s):
echo:  M203 X500.00 Y500.00 Z12.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X9000 Y9000 Z500 E10000
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S1500.00 T1500.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.20 E2.50
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:   M301 P22.98 I1.64 D80.49
echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)
echo:   M207 S3.00 F2700.00 Z0.00
echo:Recover: S=Extra length (mm) F:Speed (mm/m)
echo:   M208 S0.00 F480.00
echo:Auto-Retract: S=0 to disable, 1 to interpret extrude-only moves as retracts or recoveries
echo:   M209 S0
echo:Filament settings: Disabled

PavelSindler, 2017-02-24:
Your maximum feedrate is 25mm/s instead of 120mm/s.
Did you use precompiled hex file or did you compile yourself @chriszero ? If you compiled source code, did you use the latest configuration file? What E feedrate do you have in Configuration_prusa.h?
Is it the same if you unload filament right after resetting the printer?

chriszero, 2017-02-24:
I've used the latest precompiled .hex from github

PavelSindler, 2017-02-24:
Isn't it possible, that you have used previous version (RC1) hex file instead of the new one? Please, can you check firmware version in LCD main menu -> support?

chriszero, 2017-02-24:
It is 3.0.10,
But I've installed the 3.0.10-RC when it was released, and finally updated to 3.0.10.

PavelSindler, 2017-02-24:
Please try to run M502 (load default settings) and then M500 (store settings to EEPROM). It should solve your problem.

chriszero, 2017-02-24:
Loaded the default settings, seems to be ok now.
So there setting was from the 3.0.10-RC?

SENDING:M503
echo:Steps per unit:
echo:  M92 X100.00 Y100.00 Z400.00 E161.30
echo:Maximum feedrates (mm/s):
echo:  M203 X500.00 Y500.00 Z12.00 E120.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X9000 Y9000 Z500 E10000
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S1500.00 T1500.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.20 E2.50
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:   M301 P40.93 I4.88 D86.08
echo:Retract: S=Length (mm) F:Speed (mm/m) Z: ZLift (mm)
echo:   M207 S3.00 F2700.00 Z0.00
echo:Recover: S=Extra length (mm) F:Speed (mm/m)
echo:   M208 S0.00 F480.00
echo:Auto-Retract: S=0 to disable, 1 to interpret extrude-only moves as retracts or recoveries
echo:   M209 S0
echo:Filament settings: Disabled

PavelSindler, 2017-03-01:
Yes, it was from 3.0.10-RC. Thank you very much for cooperation.

closed 2017-03-17 by PavelSindler