prusa3d/Prusa-Firmware #48: M304 does not service reboot

geekmoto13, 2017-02-20:
M304 is not saving with the m500 this is still a bug. m301 does save however

terminal output from octoprint
`
[...]
Send: M301
Recv: ok p:18.11 i:1.38 d:59.35 c:1.00
Recv: ok
[...]
Send: M304
Recv: ok p:126.13 i:4.30 d:924.76
Recv: ok
m301 is the current tuning for the hotend
m304 is the firmware values

prior to reboot
Send: M304 P59.44 I2.43 D362.74
Recv: ok p:59.44 i:2.43 d:362.74
Recv: ok
[...]
Send: M500
Recv: echo:Settings Stored
Recv: ok
[...]

after reboot

Send: M301
Recv: ok p:18.11 i:1.38 d:59.35 c:1.00
Recv: ok
[...]
Send: M304
Recv: ok p:126.13 i:4.30 d:924.76
Recv: ok`

geekmoto13, 2017-02-20:
Fw is 3.0.10

geekmoto13, 2017-02-20:
Title should read survive

PavelSindler, 2017-02-20:
Thank you for reporting this issue.

PavelSindler, 2017-03-23:
This issue was fixed in https://github.com/prusa3d/Prusa-Firmware/commit/e7dfb98a4fbbf020bffa589e7aeb9d532a72e1a4. There are changes in variables stored in EEPROM, therefor flashing this (and newer) firmware version will lead to loading default settings (to prevent wrong data being written to the variables).

closed 2017-03-23 by PavelSindler