prusa3d/Prusa-Firmware #49: Re-add positive live Z

joeynuggetz, 2017-02-20:
I would like to request that positive live Z be re-implemented. I'm currently running a -0.30 offset and wanted to try other build surfaces but that would require I either flash back to previous firmware or play with the pinda probe.

Also, some filaments I run like PETG need less squish and I currently cannot print PETG as I typically have to add anywhere between .50 and .100 more positive offset than my PLA offsets. Not sure why this was changed. I agree that maybe negative offset should be capped so people dont scar their beds, but why lock the positive direction as it only allows you to move the nozzle further away from the bed.

If there are no plans to revert this, can someone tell me what I would need to change in the code so I can put it back the way it was?

PavelSindler, 2017-02-20:
There was no live Z limitation so far. Adding this limitation should prevent setting live Z to excessive values which can lead to heatbed damage.
Positive live adjust value means that your PINDa probe is too high and should be moved little bit lower. It is true that positive offset just move nozzle away from bed, which is safe. But if you need positive live adjust Z, there is risk that you can damage bed if you haven't run V2 calibration yet and also if you reset calibration, erase eeprom etc. 
If you have PLA offset -0.3 and for PETG you need -0.2 mm it is still ok. Or am I missing something?
I would definately reccomend moving your PINDa probe little bit lower.
Anyway, your live Z minimum and maximum values can be set in Configuration_prusa.h by Z_BABYSTEP_MAX and Z_BABYSTEP_MIN values. But this value is in steps and not in mm (so 3999 is approximately 10mm).

joeynuggetz, 2017-02-26:
Thanks for the info. The issue was my offset wasn't -0.3 but -0.030. So not a lot of headroom to experiment with filaments that require less squish and perhaps using blue take or other build surfaces for testing. I have since made the changes in firmware as you suggested and re-flashed. Thanks you.

closed 2017-02-26 by PavelSindler