prusa3d/Prusa-Firmware #55: Adding a "long pause" option

denics, 2017-03-04:
How many times has happened to you that you are printing and suddenly somebody want to go to bed and ask you: "is all this noise necessary?", kindly insinuating that you should shut down the machine and :( your creation. However, in some situation, it would be possible to simply pause and then resume in the morning.

At the moment I usually do that manually: pause, move the nozzle away from the object (or there will be some extra material under the nozzle), bring the temperature of the nozzle and the bed to 0. 

In the morning I simply restore the two temperatures, select "resume printing" and the great MK2 do the rest, sometimes you cannot even see the point where I resumed.

My point is that it would be nice to have an extra option during the printing that says: long pause and do all those tasks automatically. If you like the idea, I could even try to provide some code, with some guidance.

werewolf1479, 2017-03-04:
I think this is related to #23 and yes would be very awesome.

PavelSindler, 2017-03-06:
Yes, I also think that this feature would be great. We will add it to next release. If you have some code, please make pull request. Thank you.

closed 2018-12-11 by inquisitiveKangaroo