prusa3d/Prusa-Firmware #58: Add custom G-code to toggle high-power mode & silent mode

JohnnyricoMC, 2017-03-12:
An idea for a very minor enhancement I had this evening: a way to toggle the firmware's silent mode remotely.

Currently, we can switch between high power-mode and silent mode mid-print via the LCD screen, even while printing from USB. 

Is there a way this functionality could be replicated via a custom G-code command? (if it isn't already there but I just haven't searched enough)
I.e. sending `Mnnn 1` to switch to silent mode & `Mnnn 0` to revert to high-power mode. (where nnn is the number for the chosen M-command)

This would make it possible for users to switch between these modes remotely, i.e. via a machine running Pronterface or an Octoprint server, or even as a scheduled task.

Semi-related: The Reprap wiki describes [M913](http://reprap.org/wiki/G-code#M913:_Set_motor_percentage_of_normal_current) (only used in reprapfirmware) as a means to set stepper motor currents to percentages of their normal values, describing motor noise reduction as a possible application.

PavelSindler, 2017-03-13:
Currently it is possible to use M907 gcode in our firmware. Example:
M907 X540 Z830 E500
Current is in mA and X stands for both X and Y stepper currents.

JohnnyricoMC, 2017-03-13:
Neat. In [Firmware/variants/1_75mm_MK2-RAMBo13a-E3Dv6full.h](https://github.com/prusa3d/Prusa-Firmware/blob/MK2/Firmware/variants/1_75mm_MK2-RAMBo13a-E3Dv6full.h) I find the following values defined at lines 163 & 164
```
#define DEFAULT_PWM_MOTOR_CURRENT  {270, 830, 450} // {XY,Z,E}
#define DEFAULT_PWM_MOTOR_CURRENT_LOUD  {540, 830, 500} // {XY,Z,E}
```
Which are used in [Firmware/Marlin_main.cpp#L1957](https://github.com/prusa3d/Prusa-Firmware/blob/MK2/Firmware/Marlin_main.cpp#L1957)

I might be asking the obvious, but I take it the loud PWM motor current values are the high power-mode & the regular values are the silent mode?

This approach seems better than actually toggling silent mode (in the EEPROM), as no writes need to be done & upon a restart of the microcontroller it runs at the mode set in the EEPROM again.

PavelSindler, 2017-03-14:
Yes, you are right about the currents. DEFAULT_PWM_MOTOR_CURRENT_LOUD is for high power mode and DEFAULT_PWM_MOTOR_CURRENT is for silent mode.

JohnnyricoMC, 2017-03-14:
Thanks. I'll give it a try this evening & then close the issue.

qknight, 2018-01-26:
@JohnnyricoMC 
could we issue the `mode` related G-Codes you mentioned using something like cron so we automatically switch profiles during night/daytime. that would be awesome!

JohnnyricoMC, 2018-01-26:
I'm not sure if it's safe to send commands via an external tool when something else is already sending the printer commands. In the case of Octoprint you could set up cronjobs to send API calls to your instance: http://docs.octoprint.org/en/master/api/printer.html#send-an-arbitrary-command-to-the-printer