prusa3d/Prusa-Firmware #59: Filament change script:  Need menu option to manually unload filament if automatic unload fails.

domesticatedviking, 2017-03-13:
On my first real attempt with Colorprint and a mid-print filament change - I ran into an issue where the extruder slipped when the printer automatically attempted to unload my first color of filament. The menu options available do not cover this scenario -- I could either proceed with the print, attempt to load filament a second time, or to extrude more filament (if the first color had not been purged yet.) A fourth item would be very helpful -- one that gave the option to try to unload the filament again. Since there is no way to access the printer's main menu screens while the filament change script is running, it didn't seem that there was any way to save the print.

domesticatedviking, 2017-03-13:
Alternately / also (and maybe this is a colorprint issue more than a firmware issue) - it would be nice to have the option for the printer to signal the need for a filament change, and then wait for operator input before running the filament unload sequence.  I find that I can eject filament more reliably if i pull it out while the extruder stepper is retracting - sometimes the molten end of the filament is too wide to remove from the printer easily after it cools.

PavelSindler, 2017-03-13:
Thank you for your feedback.

darkvertex, 2018-02-14:
Hi! Any update on this? Every time I use ColorPrint and come back to my beeping printer, the filament has slipped back in and my choices are feed in further or feed in further. :(

I then proceed to do a dance of confirming then immediately pausing from my Octoprint control server then sadly it pauses while above the part being printed, often drooling a thin glob of wrong color filament to it that I have to peel off by hand after manoeuvering the printer head to go up a bit and to the side so I can then manually unload and reload the proper filament from the LCD menu and finally tell Octoprint to resume.

It's super inconvenient and often ruins my print cause I can't always pull the filament drool off in time.

Just needs a "Filament did not unload, please try again" option! Please.

3d-gussner, 2020-07-13:
@domesticatedviking This issue has been stale for long time, can you please update it or if it is already fixed with newer firmware version close it.
[stale]

JohnnyDeer, 2020-08-12:
I'm closing this issue as stale, thanks for understanding.

closed 2020-08-12 by JohnnyDeer