prusa3d/Prusa-Firmware #61: Prusa3d Specific G-code explanation?

sarf2k4, 2017-03-15:
Hi, I would like to know what do these g-codes do when issued especially

- G80
- G86
- G87
- G88

From my point of view at G80, it is almost similar to G29, but another addition after all 9 points has been probed is that the print head will go to position 0,0,0. I need confirmation on G80 though because at the start g-code of slic3r, no Z movement were issued.

PavelSindler, 2017-03-17:
G80 measures height of all 9 points (relative to zero point), apply live adjust Z and apply bed level correction on measured points. Then it counts correction matrix which is used in planner for Z coordinates correction when printing. 
G86 and G87 changes calibration status. Calibration status is just information about calibrations which have been run successfully. This infromation is stored in EEPROM. If some calibrations have not been run yet, user is informed what is the next step. G88 is reserved for future use.
Using G86 and G87 in your gcodes can set calibration status to value which doesn't correspond to calibrations which have been run.

sarf2k4, 2017-03-17:
Does G80 will move to the 0,0,0 position after it has probed all 9 points?

PavelSindler, 2017-03-17:
Yes it does.

sarf2k4, 2017-03-17:
Another one, basically it is a G29 but added extra path to move xy first, then z to 0 lastly. 

I can use the old G29 instead of G80 also right?

PavelSindler, 2017-03-17:
Why would you do that?
G29 is not used in current firmware version. If you send G29 via serial, nothing happens.

bubnikv, 2017-03-17:
http://reprap.org/wiki/G-code

G29: Detailed Z-Probe

G80: Cancel Canned Cycle (CNC specific)

There has been a long discussion on various forums, that we should have
used G29 for the bed leveling. First, our bed leveling does a bit more than
what G29 is supposed to do, second the decision has been met a year ago and
now it is late to change to G29.

closed 2017-03-22 by PavelSindler

fbujold, 2017-10-04:
I do not understand the difference between G28 and G80