prusa3d/Prusa-Firmware #63: Home Before Calibration

sarf2k4, 2017-03-19:
One thing I noticed is that there is no auto home sequence added right before either calibration option xyz nor z.

Without auto homing before this, this also increases the risks of having the nozzle crashing onto the bed since the probe aren't always on the probe point unless manually homed first

PavelSindler, 2017-03-20:
There is no need to manually home before xyz calibration nor z calibration.
Have you observed some issue regarding this?

sarf2k4, 2017-03-20:
If i remember correctly, the printer asked me to raise the z axis directly
without homing, then lowers down. I dont know until when the axis will stop
moving. I eventually will do check again later

sarf2k4, 2017-03-22:
okay, I've confirmed that even without homing before doing the calibration, the printer will stop at that particular height. This actually scares me a bit especially when it got close to bed

closed 2017-03-22 by sarf2k4