prusa3d/Prusa-Firmware #64: Bed Correction Resolution?

sarf2k4, 2017-03-22:
Hi, I stumbled across this thread http://shop.prusa3d.com/forum/software-f13/i-need-help-printing-abs-please-its-driving-me-mad-t3416.html and I got to agree that the bed correction resolution are somewhat coarse and can't really be used if either bed corners are high or low.

I also would like to suggest to increase the bed correction resolution, by using probe points as index number and reference, adjustable to a max of about +-100microns. The probe point index will be based on the auto bed leveling index point, meaning up to 9 points can be adjusted

PavelSindler, 2017-04-19:
It is interesting idea. Adjusting 9 probe points would be not very helpful, because mesh bed leveling (9 point calibration) directly measures Z coordinate in these points.
In some cases (more complex shape of heatbed) measuring in 9 probe points could be insufficient for precise first layer. You have very accurate Z coordinate in probe points, but you don't know what is between probe points. So adjusting Z coordinate in areas where you don't have probe points could be more helpful.

sarf2k4, 2017-04-20:
yes, i am quite aware of the heatbed shape, in the main marlin branch they are still working on the Uniified Bed Leveling, which will take samples of the surface point then we got to specify the z height for that. After that, the marlin will eventually adjust the z height according to the adjustment made.

What I had in mind for these values is that similar to the one being used now, since there are two types of variable being used, z-offset (used in the live z adjust) and bed correction. I'm pretty sure that the current bed correction in the prusa firmware is like this

z-offset - bed correction = final adjustment

I don't mind if it's not directly on the probed point though, just having higher resolution would help especially in the corners, who knows if the bed itself are not flat or bow shaped which is common

PavelSindler, 2017-04-21:
Yes I understand what you mean. Thank you for your feedback.

Ludio, 2017-05-22:
So I was thinking about how depth finders on boats work by sending a signal and reading the echo. They also have the ability to show topology. The reality is that PEI bed is not 100% flawless and becomes more flawed as time goes on. The adhesive allows the surface to flex (or bulge) and over time will level again after printing in other areas. It would be great if we could use some form of distance scanner to actually scan the PEI imperfections into a two dimensional array. This would help when you have the Z height just right but board imperfections throw off the first layer. I've see the PEI flex enough to cause adhesion issues is areas which were once flat but have changed due to printing.

Perhaps something like this but I am not sure https://www.zoro.com/omron-photoelectric-sensor-rectangular-diffuse-e3z-d61/i/G2320866/?gclid=CJ-i_YKagtQCFU1ufgodG20Jqw

closed 2018-12-11 by inquisitiveKangaroo