prusa3d/Prusa-Firmware #67: Version check integrated into M115 nuking communication over serial

foosel, 2017-03-24:
I just had a user run into [this issue](https://github.com/foosel/OctoPrint/issues/1831): Print job containing a `M115 U3.0.10` (injected by the slicer, I guess it's even your own Slic3r version) effectively nukes serial communication. A look at your firmware source reveals [this line](https://github.com/prusa3d/Prusa-Firmware/blob/12505d0f4ed588b66571ecac94b17c4e2f35a6f2/Firmware/Marlin_main.cpp#L4174) causing the serial comms to be stalled while user gets shown a message to upgrade the firmware, introduced in 0b372bfc946a76619bb60fb66d28489f2f933eac. That's really really bad because external hosts do not have any way to abort this or even to know why the printer stopped replying. The printer is effectively dead on the serial interface and the only option is a disconnect (or the user has to be physically present to handle the LCD prompt).

I understand the reasoning behind this behaviour (being able to inform the user that the GCODE cannot be run on the current firmware version), but the current implementation causes a huge headache for host software like OctoPrint.

Could you please look into a solution that isn't as disruptive?

In the specific case of OctoPrint, returning an actual "Error: Upgrade your firmware" or something like that on the serial line (while also displaying a message on the display maybe?) would be way more straightforward, since it would allow me to cancel the (incompatible) print job and shutdown the connection gracefully, instead of just freezing up. Like wise for the use of an [action command](http://docs.octoprint.org/en/master/features/action_commands.html).

As a general rule of thumb, any kind of blocking LCD-only behaviour triggered through commands received via serial is really bad for interoperability and a nightmare for host software devs.

PavelSindler, 2017-03-24:
Thank you very much for detailed description of this issue. We will improve it.

ayourk, 2017-03-31:
The Calibration XYZ and Calibrate Z are some other instances of serial blocking behavior.  Some other cases where the serial line is blocked is when the printer starts up and hasn't been calibrated of the Live Z is not set.

foosel, 2017-03-31:
Don't know how it behaves now, but the `M600` filament change is also usually a prime candidate for this offense.

Some/newer versions of Repetier Firmware send `echo: busy process` when a longer process is in progress (regardless of LCD interaction I believe). That at least allows the host to know "ok, something is happening, printer is still there but busy, nothing I can do right now". Still, it might make sense to establish some regular heartbeat like message more specific to LCD interactions (`echo: action required`?) that hosts could parse for and hint their users at that their printer demands their physical presence. That would need to be coordinated across major firmwares to make sense though (and past experience has shown that this is just as big of a nightmare sadly).

Anything is however better than complete silence on the serial line during blocking actions, since that is indistinguishable from a crashed controller or some versions of yanked out cables.

foosel, 2017-07-28:
`M600` not producing *any* output at all while the printer is waiting for user interaction on the controller is causing issues for users in the field, see foosel/OctoPrint#2026 just as expected. 

*PLEASE* look into implementing the `busy` protocol present in both current Marlin mainline and Repetier firmware variants, it solves exactly these kinds of problems.

I can't work around this issue from the host side, I can't distinguish a printer not replying anymore due to a blocking controller action from a printer that's crashed.

bubnikv, 2017-07-28:
Thank you Gina for heads up. We are actively discussing this topic in our RnD group and it is important to us, that our printers work well with your OctoPrint.

Is there any documentation of the busy protocol available?
Do you have any other suggestions for improving the communication protocol?

foosel, 2017-07-28:
Not much in terms of documentation. It boils down to sending 

```
echo:busy: processing
```

back to the host every n seconds (I think Marlin defaults to 2) if the printer currently doesn't accept new commands due to a blocking command/is busy doing something in its command loop and doesn't handle any new input. As far as I know that is implemented in the idle loop that is also taking care of managing the heaters and steppers.

There's also

```
echo:busy: paused for user
```

in case of a blocking user interaction (like `M600` or anything else where the user is expected to interact with the printer via its controller).

See also [this PR that added the functionality to Marlin](https://github.com/MarlinFirmware/Marlin/issues/3109) and [this initial issue on the Marlin bug tracker from me on this topic](https://github.com/MarlinFirmware/Marlin/issues/3068).

From OctoPrint's perspective, *any* kind of regular output on the serial line while the firmware is blocked and busy doing something already is enough to avoid timeout issues. So whether you send `echo:busy: paused for user` here or `echo:busy: processing` or `waiting for a button press` or `.` is irrelevant as long as you send *something* at regular intervals while blocked. Of course, sticking to something that other firmwares are already doing is in general a good idea - we do not have a well defined standard in place, so we need to stick together.

This is also not just an OctoPrint problem btw ;)

bubnikv, 2017-07-28:
Thanks Gina for the details.

Naugrimohtar, 2017-08-14:
I have experienced this issue with M600 printing while away expecting to come back with my printer waiting for me (which it was) but OctoPrint timed out and now my print is wasted. This first happens running guide from my pi then again from SD card via OctoPrint. Now that I have found this bug, I understand the cause. I am hoping for a fix!!

GurliGebis, 2017-08-14:
I have an initial pull request for this, it is not working as expected though :)

closed 2018-12-28 by inquisitiveKangaroo