prusa3d/Prusa-Firmware #71: DOGLCD implementation issue

moranelli, 2017-04-03:
Hi, Sorry for my English.
I am trying to compile the firmware version 3.0.10 for use with the lcd display: http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller ,
but I get compilation errors. In Marlin versions this works correctly, 
but in firmware 3.0.10, when setting up in the same way that in Marlin the file dogm_lcd_implementation.h is incomplete.

Configuration firmware 3.0.10 for Prusa i3 MK2

******************************************************************
//#define ULTRA_LCD  //general LCD support, also 16x2
#define DOGLCD  // Support for SPI LCD 128x64 (Controller ST7565R graphic Display Family)
#define SDSUPPORT // Enable SD Card Support in Hardware Console

//#define REPRAP_DISCOUNT_SMART_CONTROLLER

#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

******************************************************************

When using REPRAP DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER option the process defines DOGLED using the file dogm_lcd_implementation.h
#if defined (REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)
 #define DOGLCD
 #define U8GLIB_ST7920
 #define REPRAP_DISCOUNT_SMART_CONTROLLER
#endif

The following logic is in the ultralcd.cpp file
#ifdef DOGLCD
#include "dogm_lcd_implementation.h"
#else
#include "ultralcd_implementation_hitachi_HD44780.h"
#endif

When using the library dogm_lcd_implementation.h the compilation process fails because several functions are missing, such as:

lcd_implementation_init_noclear
lcd_set_custom_characters
#define lcd_implementation_drawmenu_back_RAM_selected(row, str, data) lcd_implementation_drawmenu_generic_RAM(row, str, LCD_STR_UPLEVEL[0], LCD_STR_UPLEVEL[0])
#define lcd_implementation_drawmenu_back_RAM(row, str, data) lcd_implementation_drawmenu_generic_RAM(row, str, ' ', LCD_STR_UPLEVEL[0])

#define lcd_implementation_drawmenu_setlang_selected(row, pstr, data) lcd_implementation_drawmenu_generic(row, pstr, '>', ' ')
#define lcd_implementation_drawmenu_setlang(row, pstr, data) lcd_implementation_drawmenu_generic(row, pstr, ' ', ' ')

All functions are defined in the library ultralcd_implementation_hitachi_HD44780.h

If I comment the line "#define DOGLCD" the compilation process uses the library ultralcd_implementation_hitachi_HD44780.h,
The process compiles correctly, but nothing appears on the display.

I wanted to know if there are any options that I am missing comment or uncomment.

Thanks and Regards.

wschadow, 2017-04-15:
Hi, thanks for reporting this! I have exactly the same issue. I can compile Marlin software for the RepRapDiscount_Full_Graphic_Smart_Controller, but the Prusa modified version. I get the same errors. It would be great if you could fix this. For me, this is a downgrade of the software, when this display is not usable anymore.

bladykast, 2017-05-20:
Same exact issue. I have found a way around most of the implementation issues, but I am stuck on errors like "dogm_lcd_implementation.h:486: error: 'lcd' was not declared in this scope". I don't know if it'd be a better option to try and run Marlin's version of ultralcd.cpp and dogm_lcd_implementation.h and try to get Prusa's mods working.

das300hub, 2017-05-21:
A couple of issues appear to me . I also am in this situation doing a conversion from an old i3 to a mkll prusa project. So far everything has been manageable with hardware problems solved up to the point of calibration and operation.
Note 1 : the prusa software only calibrates from lcd which prevents people like me calibrating as i own a doglcd not the generic prusa version. It would be prudent if either the lcd issue of connecting a doglcd was fixed or better if a multi platform menu system could be run via usb and a pc or via a web browser.
Note 2: the prusa runs minirambo and i am not completely sure the hardware is identical to the ramps and mega combo. If it isn't this might explain the issues after compiling. I briefly checked schematics of both lcds to see the differences but whats required is a decent hardware check of
- minirambo versus mega and ramps1.4
- differences between lcd types and connections required.
Note 3 : when connecting doglcd 128x 64  display the encoder push button seems responsive as does the cd card check pass in prontoface . This hints that the sd card wiring may be similar . The lcd though does not work naturally as prusa software in minirambo still thinks it is for the generic lcd. I have not tried compiling anything from standard , hence why i am here reading this post.

It seems hardware needs to be checked first so perhaps we can share links to the schematics for ramps , minirambo , lcd types and then establish what hardware difference are present. This may then tie in with explaining why  software still doesnt operate after compilation.

I hope at that stage we can come up with a solution instruction so others can get calibration outside of using the standard lcd.

Also, if anyone is interested in creating a menu driven from a pc please let me know as i would be interested in solving this part of the problem. I think its worth jo prusa getting involved with that issue because it opens up more business for people wishing to convert their systems into a prusa style unit who cannot afford a full system .

michalxfanta, 2019-04-11:
This issue is outdated.

closed 2019-04-11 by michalxfanta