prusa3d/Prusa-Firmware #72: SD card statistics recording

lmcnjmiller, 2017-04-11:
When a print is done via SD card a log file is generated that says how long a print took and perhaps other information. This would be helpful in keeping overall statistics and keeping track of multi part prints.

m-Th, 2017-06-29:
How much time - ok. 

But we need more info in this log in a tab separated format [Date/Time stamp][Tab][Event]
Events (from top of my head):

- Print Start
- Print Pause
- Print Resume
- Print Stop
- Load Filament
- Unload Filament
- Change Tool
- Live Adjust Z
- New Layer
- Speed Manual Change
- Nozzle Temp Manual Chg
- Bed Temp Manual Chg
- Other information: How much filament was used between Load/Unload Filament commands. 

A Printer Reset (the „X” button on the front panel) should write flush the log on the SD card with the last event being „Printer Reset”

In fact, Marlin has a G-Code command (M928: Start SD logging) but I don't know what it writes in that log...

lmcnjmiller, 2017-06-29:
Looks like M928 logs all g code commands given to the printer until M29.

m-Th, 2017-06-29:
This is what I am afraid of. Too much informational noise.

closed 2018-12-11 by inquisitiveKangaroo