prusa3d/Prusa-Firmware #73: Move axis during leveling

sarf2k4, 2017-04-12:
Hi, I am wondering whether or not it is intended to allow the move z axis during leveling (non-printing execution through lcd panel).

I noticed it this morning when I kind of fiddling with the lcd panel, decided to do mesh bed leveling followed by move z axis to 100mm. What I did is to initiate leveling procedure and during the process I raise the z axis. To my surprise was, the z axis corresponds to the move axis command issued during the procedure.

I would like to address this behavior whether it is a bug or not. If it is a bug, I would like to suggest that the leveling procedure will be completed first before executing the other command

fiveangle, 2017-04-13:
I would think a manual move z should always be honored, immediately abandoning the bed leveling procedure, with an LCD/serial error indicating such.

sarf2k4, 2017-04-13:
To be precise, it didn't abandon the leveling procedure, it is just moves up and down during the procedure =\

closed 2018-12-11 by inquisitiveKangaroo