prusa3d/Prusa-Firmware #74: Fix for a wrong checksum truncation for certain commands

mnl1121, 2017-04-14:
Please add this fix to the firmware!
https://github.com/MarlinFirmware/Marlin/commit/2d22902d080b08c65ebdf5f8b3f03529ccd58144

Without it OctoPrint can't be used to print from the web interface as uploading, selecting, starting files doesn't work. 
http://i.imgur.com/5B4xcyh.png

mnl1121, 2017-04-14:
Hmm looking at the code it appears to be already implemented. I'm not sure why then, but the file names are incorrect when viewing on OctoPrint. Also can't upload, select, or start printing any files using OctoPrint.

bubnikv, 2017-04-14:
The FAT library which the Marlin integrates does not support long file names. In general, the FAT code does not support manipulation of files by their long names. The only place where the long file names are supported is the directory listing,

mnl1121, 2017-04-14:
Can you explain further? You mean it only supports short file names as in 8 characters or fewer? For OctoPrint to be compatible do I only need to ensure the file name is short enough?

bubnikv, 2017-04-14:
In the times of MS DOS the file names had maximum 8.3 (8 for file name, 3 for extension) characters. Then came Windows 95 with a VFAT file system supporting long names, but also maintaining a legacy 8.3 file name for each long file name. These 8.3 short file names are still supported and you can list them by running dir /W on Windows.

So you can work with the long file names by their 8.3 counterpart.

mnl1121, 2017-04-14:
I see. So this problem is unrelated.

closed 2017-04-14 by mnl1121

mnl1121, 2017-04-15:
For anyone else with the problem of not being able to upload, select, or start prints, take out the SD card and try again. Upload directly to the Pi instead and OctoPrint works great with the Prusa i3 MK2.