prusa3d/Prusa-Firmware #75: Linear Advance Integration

Quintox303, 2017-04-15:
I've noticed that the firmware still refers to the old version of the advance algorithm in Marlin, and is not able to be configured through M905 like it is now in stock Marlin fw. Refer to http://marlinfw.org/docs/features/lin_advance.html. A new LIN_ADVANCE feature has been introduced which has been shown to drastically improve print quality in some areas.
It would be great if this could be integrated in the next update

PavelSindler, 2017-04-21:
Thank you for this information.

bubnikv, 2017-04-21:
@quintox303 By the way, do you know whether it works and how well does it
work?
I know there is an implementation of the pressure advance in Repetier and
Sailfish and AFAIK it only works for Sailfish. I did not follow the
development of Marlin since the last summer though.

Quintox303, 2017-04-21:
@bubnikv Yeah I've seen evidence to show that it definitely does work - and really well too! There was a discussion of it at this post on the facebook group https://www.facebook.com/groups/Prusai3users/permalink/732860686898415/. The user posted pictures taken from another group of prints done on a tevo tarantula (bowden extrusion). If you have a look, you can see the prints are perfect. 

There have been suggestions that pressure control isn't relevant for direct drive systems or that it wouldn't make a difference. But the pressure control doesn't only account for spring action in the feeding system before it reaches the nozzle, but also for the actual pressure build-up in the nozzle before extrusion begins. Hence it would be applicable for all extrusion systems. Also the testing that was done for the original documentation I posted in the link of my original post, was all on a direct drive printer.

I also posted some pictures in the comments of that post showing the bulging corners that I was experiencing - something that can and would be fixed through the use of pressure control

r0ndL, 2017-04-21:
Would love to see this enabled as well.  I must reiterate Quintox303 statement that is **beneficial to direct setups as well** not just bowden.  Additionally while "slicers have options to control the nozzle pressure in some ways. Common names are: Pressure advance, Coast at end, extra restart length after retract. Disable them all! **The firmware will do a much better job than the best slicer can do**, and they would influence themselves in a bad way. It is also strongly recommended to disable further features like wipe while retract.

fiveangle, 2017-04-21:
I have 1st hand experience that it makes a **huge** difference in allowing print speeds that are otherwise unattainable without it.  I am considering switching to Marlin 1.1 when it releases to gain this functionality on my MK2, even though it would mean losing all of the wonderful benefits of the PR stream.

On a bone stock _Printrbot Simple Metal with Heated Bed 1403_ with Linear Adv enabled and tuned for PLA I was able to attain identical quality at true 70mm/s that I had to drop to 30mm/s to attain without it.  Any model with lots of sharp corners benefits greatly by not requiring glacial-like speeds to keep excellent print quality.  The big benefit I see for the PR i3 is the ability to print the more complex but strong Cubic infill at probably double the speed as what we can now, with zero degradation in print quality and strength.

If you print at 20mm/s it doesn't gain you anything, but printing models in an hour at 80mm/s for "Fast" rates with better quality than a 2.5hr print without it really improved the satisfaction level of my Printrbot for prototyping phase of model design.  Nobody cares about speed for final rendition, but that's perhaps only around 15% of the prints we do, right ?

It would be oh so grand if PR ported this feature to the i3 stream with Release Notes claiming it "Experimental" and have the "K" factor set to zero by default, which would not change behavior from current, but those of us who understand how to use the feature would gain a 2 to 4-fold factor improvement in "Fast" print speed.  Once in the stream, your team could begin kicking the tires of how to best incorporate it.  I could envision the already fast "Fast" Slic3rPR profile print time being cut in half (no, I'm not kidding... although you may have to drop the LH down to 3, since 3.5 is pushing the physical limits of the 0.4mm nozzle).

With the L.A. feature, it would really help bridge the gap for the PR i3 between a great consumer printer, and a great prosumer printer.

fiveangle, 2017-04-21:
I forgot to mention: top-layer is dramatically improved even at normal speeds.

Regardless, below are two models printed with and without at 120mm/s with 800mm/2^2 acceleration values on PBSM1403. Notice the uncontrolled bulging of the fingers around the ring.: [image]
[image]

(Original model by wstein: http://www.thingiverse.com/thing:598964)

GurliGebis, 2017-05-17:
Looks like there is a PR for this awaiting :)
https://github.com/prusa3d/Prusa-Firmware/pull/93

psavva, 2017-05-31:
@Sebastianv650 is the developer for LIN_ADVANCE
Awesome prints with it enabled.

bubnikv, 2017-05-31:
I was looking into the pressure advance theory and implementations since I joined Prusa Research the last spring. I certainly see a value in it and I was toying with the idea of porting the Sailfish pressure advance to Marlin, but due to a lack of time it never materialized. I checked the implementation by @Sebastianv650 this Monday and I find it very clever and cleanly integrated. I especially like the idea how the E events are interleaved with the XY events sharing the same interrupt.

Sebastianv650, 2017-05-31:
Thinks @psavva for informing me about the discussion here :)
I had a quick look at the postings happened before. Regarding the topic direct drive vs bowden, I would even say `LIN_ADVANCE` is better suited for direct drive systems than for bowden. This is due to the fact how Marlin (and I guess most of it's forks) handle the extruder speed / jerk. While it's possible to get smooth (=no speed jumps) movements out of 3 axis, this is not true for a 4th axis (the extruder) except you allow a full stop at each corner. Think about extrusions at different width inside one outer perimeter. Each speed jump will be amplified by `LIN_ADVANCE`. On a direct drive system, the needed K values are "low" so the extruder will likely be able to follow the movements. On a bowden system with a "high" K value the extruder might skip steps. Therefore, `LIN_ADVANCE` might not be useable on every printer and every filament out there.
Beside that, on a bowden system there are friction effects due to buckling inside the tube, and I have no idea how it might be able to compensate them.
Note that I don't own a bowden printer, so I have only reports of other users and physical knowledge as a base for the text above. I use a geared direct drive printer with PLA only (except some special prints).

markaswift, 2017-06-16:
+1, can't wait for this!

Sebastianv650, 2017-06-16:
I'm currently working on the porting to the Prusa i3 MK2 Firmware. Up to now I don't have an i3, so all my coding is untested and I can only check if it "looks good" and compiles. But that should change in the near future :)

I will keep you updated if there are news and of course provide a link to my Github Fork as soon as I think it is ready for beta-tests.
Nevertheless please be patient, it's not only porting. First I have to get, build, calibrate and get familiar with the new i3, after that the first LIN_ADVANCE code tests will start. If I start directly with my modified FW, I will never know if something happens due to a bug introduced by my changes to the FW or if it's a fault of calibration or a build error.

Quintox303, 2017-06-16:
Sounds great! I'll be happy to test it as soon as a beta is ready

bubnikv, 2017-06-16:
We at Prusa Reserach are excited as well :-)

GurliGebis, 2017-06-16:
@Sebastianv650 Could you take a look at the work done here, if that would be of any use to you? https://github.com/prusa3d/Prusa-Firmware/pull/99
Currently, it seems like it won't boot with the changes in there (I have only separated it out from the other pull request, that included other changes).

Sebastianv650, 2017-06-16:
Oh, I wasn't aware there is already a PR! But if it's not working, it's maybe a good idea anyway to start with my own thing.

On a first look, I would state the following:
- Stepper.h defines Ex_DIR_WRITE and related methods only up to 3 extruders. In stepper.cpp, you allow also the call of E3, which will not exist.
- If this PR leads to a failed startup (no display), it's most likely due to a problem with ISRs. This PR does two steps at once (if not more): My concept of allowing specific ISRs inside some others and LIN_ADVANCE itself. Both manipulate the ISR behaviour, that's why my first implementation does not contain the first one (allowing interrupted ISRs).
- Inside stepper.cpp, you are changing setting of Timer0. I Don't know whats the meaning of this. LIN_ADVANCE is using only Timer1 as the default stepper ISR is using it also:
```cpp
#ifdef LIN_ADVANCE
  #if defined(TCCR0A) && defined(WGM01)
    TCCR0A &= ~(1<<WGM01);
    TCCR0A &= ~(1<<WGM00);
  #endif
  e_steps[0] = 0;
  e_steps[1] = 0;
  e_steps[2] = 0;
  TIMSK0 |= (1<<OCIE0A);
#endif //LIN_ADVANCE
```

Maybe this helps if you want to try to fix it.

My version of LIN_ADVANCE for Prusa is a slightly stripped down version. Bubnikv asked for a simple code to ensure usability, and I think that's possible as this FW doesn't has to support every possible printer configuration. Therefore I removed all the extruder related arrays from my variables. On an i3 (even with multi color) there will be never the case where 2 or more extruders operate in paralel for printing. Therefore we will always have a single active one.
As far as I got the multicolor thing up to now - might be wrong - it will always use "the same" stepper, only switching the real stepper over the expander board. Another reason for eleminating arrays. On the other side, when it comes to multi color version of the FW I think we have to add some lines of code to tell LIN_ADVANCE to reset all values on a color (=stepper) change. We will always start with "0 pressure", but that's something for a later time.

A more important decission has to be made how K will be provided to the printer for Prusa FW. My original implementation wasn't storing K into EEPROM, just because it's not a fixed printer value but related to the filament type, which can change.
If the user mainly, but not solely, uses one material kind and switches to another after a longer time there is a high chance he will forget to change or calibrate K. K should be something like the nozzle temperature, you have to select it on every print start and especialy also when you change material during a print (again thinking about multi color/material).
Therefore one option would be to add a specific K to the material preheat settings. Two points are against this:
- The user can't change this settings except by flashing a new FW.
- This settings are not called when the slicer issues a filament / material change.

As a solution, I would recommend to implement an option to store K values filament based inside the slicer. This could be done by a filament change gcode script or (even better) automaticaly by Prusa Slic3r edition. This way, K would be set over M900 on every print start and filament change, in the same way as the slicer has to set the nozzle temperature at the same time point.

GurliGebis, 2017-06-17:
@Sebastianv650 The PR I have is a cleanup of another PR, that contained some other stuff as well, that did not make sense to merge in at the same time.

Looking at your feedback, it is clear that there are several other issues with it, so I'm all for going with your work, especially since you know alot more about both the firmware and LIN_ADVANCE :)

I'll keep an eye open for your code, and give it a try when it is ready (and I find out how to recover from a bad firmware flash - not going to experiment with that without having a backup plan)

fiveangle, 2017-06-17:
@Sebastionv650 I never understood why baseline Marlin includes a non-zero K value. Was there a reason it was chosen ? I of course set it to zero at compile then use gcode based on filament (or default "0") for the bulk of filaments that I haven't tested. I don't know how anyone would use it otherwise.

Sebastianv650, 2017-06-18:
@GurliGebis OK then let's keep the existing PR as a reference.
What do you mean with a bad firmware flash? As far as I know it's nearly impossible to damage the Bootloader because the firmware will not write into this memory area. Therefore even if you flash a completely broken FW version, just load the original one in Arduino and flash again. I also managed it one time to flash a stock Marlin (without my configuration) to my printer with the result that the printer was completely unusable ;)

@fiveangle you are right, especialy if we assume the Slicer should set the K value with the start-gcode script a default 0 would be much better. I will propose this as a default for the Prusa version.
The reason behind the non-zero default value was to give users a feeling for a possible valid range of a K value. Without any hint, it would be hard to find out if you should start the search during calibration around values 0-10 or 100 or even 10000. But this realy should go into the description inside the comments!

GurliGebis, 2017-06-18:
@Sebastianv650 If you flash the wrong hex file, the bootloader will be gone, and AFAIK the you have to use the ICSP header to reflash the bootloader back into the atmega chip :)

Sebastianv650, 2017-06-25:
A small update for the interested ones:
I got my i3MK2 kit this week and built it yesterday. After little calibration, I got my first test cubes this morning :)
After tracking down an ungly copy & paste error in my LIN_ADVANCE i3 test FW, it's now basicaly "working". Working means all axis are moving and it just printed a quick K testcube. More testing to be done now!

nrepian, 2017-06-25:
Nice work @Sebastianv650 :). Have you got any photos you could share with it enabled vs default? I'd be interested to see how it looks on the MK2S.

Sebastianv650, 2017-06-25:
I took two test peaces and made a short presentation. In reality it's even more impressive, but the pictures should do it:

[First LIN_ADVANCE test result.pdf](https://github.com/prusa3d/Prusa-Firmware/files/1100335/First.LIN_ADVANCE.test.result.pdf)

thess, 2017-06-25:
@Sebastianv650 - This is super! It would be nice if others could try it. At the moment, the firmware in your repo compiles OK, but the extruder is in-operable. Could you push your current version so I can give it a go? TIA

Sebastianv650, 2017-06-25:
I know my Git is broken at the moment. But please give me one or two more days for testing and code cleanup and completion. I want to have a quite solid thing I can give to all of you that want to test it before the PR, not a alpha-draft which only generates dissatisfaction and error reports.
I also want to find the perfect K for testers, but 50 seems to be quite good as a first shot.

bubnikv, 2017-06-25:
This looks great, thanks! We are looking forward to test it on our print
farm :-)

nrepian, 2017-06-25:
Really nice results - very noticeable improvement. I'm looking forward to giving this a try :).

markaswift, 2017-06-25:
Great improvement, although minor to my eye... That said, it's an advancement towards perfection! :)

fiveangle, 2017-06-25:
> This looks great, thanks! We are looking forward to test it on our print
farm :-)

@bubnikv - you may find the parts don't meet your specs due to all the fine tuning you've put into tweaking your farm for production.  You've surely tweaked inputs to perfection in order to compensate for the limitations without LIN_ADVANCE in order to get the desired output.  I found the increased dimentionsal accuracy with fast prints using LIN_ADVANCE caused me to have to go back and tighten up dimensions that mate to square-cornered parts, decrease hole size for self-tapped thread inserts, etc.

So be sure to test actually assembly of the parts before shipping to MK2S users ;)

But after recalibrating your process, I suspect this feature will increase your farm output at least 50% at better quality.

probonopd, 2017-06-25:
> I suspect this feature will increase your farm output at least 50% at better quality

Interesting. Which settings do you suggest to change (e.g., which speed settings to increase, and typically by how much)?

fiveangle, 2017-06-25:
Typically, all of the tweaking to reduce speed from max for different activities (perimeter, outside perimeter, infill, top layers, support, etc become relatively moot since LIN_ADVANCE allows more accurate deposit of the amount of material under wider range of conditions.  That said, I suspect the same speed changes may help at the new upper top-end but would have to be determined by trial-and-error.

For me, using Simplify3D on an otherwise stock Printrbot Simple Metal 1403 with Heated Bed, Where it required 50mm/s max with 1500mm/s^2 acceleration plus various slow-downs to attain perfect quality, I found I was able to remove all slow-downs, increase max speed to 100mm/s, decrease acceleration to 800-1000mm/s^2 and print and usually 50-70% faster with higher quality end result.

In short it broadens and extends the range of print speeds that allow quality results so we don't have to hand-tweak every little thing.

Sebastianv650, 2017-06-25:
@markaswift On one side it looks slightly more impressive in reality, each picture is always 2D so something is lost. On the other side, it' how you define minor and major.
The final thing that made me thinking about starting LIN_ADVANCE development was the print of an extruder body with PETG. I wanted to print it quite fast, there was no need to have a nice looking part. But I always get gaps between the perimeter lines, as we can also see them on the pictures in the pdf. This means the part is defect for me - it will fail during load much earlier than a print without gaps.
Therefore, on (and for me the biggest) advantage from LIN_ADVANCE is that we can now print much faster without voids somewhere inside the print, resulting in mechanicaly full functional parts.

When it comes to visual quality, LIN_ADVANCE also helps with that especialy on mechanical parts with corners. As the red lines in my pictures show, a cube has not a cube shape but something like I would call a pillow. This is a visual problem at first, but can also be a problem when it comes to mechanical fit of parts as @fiveangle stated above. Another thing is top surface quality, mostly noticeable on parts with flat top layers where you get a quite rough infill to perimeter intersection area without LIN_ADVANCE.

Of course, slowing down print speed is a solution. But why print external perimeters at 20mm/s when our hotend is capable of extruding up to 10mm³/s and our axis can to above 100mm/s? With LIN_ADVANCE, you can get the same quality at much higher speeds.

@probonopd it's not possible to say you can increase speed up to x times compared to without LIN_ADVANCE, it depends on the parts and your needs. But on my TAZ, I sometimes just set all the speeds to 60mm/s without slow downs with great results. An upper limit (at least with Marlin on my TAZ) is given by the calculation speed of the Rambo board - when it starts stuttering in arcs, it's too fast and this harms visual quality much. I looking forward the see of something happens also on the i3, but my first curvy print today was no problem at 60mm/s so it looks to work better :)

bubnikv, 2017-06-25:
@Sebastianv650 I did some optimizations of the planner code in November
2016,
when I converted some of the planner code from floating point to a fixed
point arithmetic.
Also it helps, that our printers are equipped with plain character displays
as serving a large
graphics display is quite CPU consuming. So it may be, that our firmware
will start
stuttering at quite higher print speeds than the TAZ.

Sebastianv650, 2017-06-26:
I think it's ready to be released to everyone who want's to give it a try. I rebased my branch and added last bugfixes and cleanups:
https://github.com/Sebastianv650/Prusa-Firmware/tree/MK2-LIN_ADV-DEV

Feedback is welcome!

Sebastianv650, 2017-06-26:
Here is a flat cube for K calibration tests, also see http://marlinfw.org/docs/features/lin_advance.html for more details. M900 Kxx is used to set a K value.

[Prusa_K.txt](https://github.com/prusa3d/Prusa-Firmware/files/1102587/Prusa_K.txt)

totalitarian, 2017-06-26:
Very nice. What firmware version is this based on?

Sebastianv650, 2017-06-26:
It's based on the Prusa MK2 branch as it is today.

Swiftnesses, 2017-06-26:
Would someone be so kind to compile a hex?

Sebastianv650, 2017-06-26:
Here you go, created using Arduino. It's for a i3 MK2 with 1.75mm filament and Rambo 1.3 - no multi material. Use at your own risk!

[Prusa i3 MK2_LIN_ADV.txt](https://github.com/prusa3d/Prusa-Firmware/files/1102762/Prusa.i3.MK2_LIN_ADV.txt) (rename to .hex)

GurliGebis, 2017-06-26:
Nice, with a bit (a lot unfortunately) of luck, someone with commit access will see it and merge it. Would be nice to have in the firmware though :-)

Sebastianv650, 2017-06-26:
I will create a PR when there are some positive feedbacks. Nothing is worse than a quick untested PR that might also get merged as a worst case ;)

totalitarian, 2017-06-26:
I was hoping it would be the latest RC branch as as was having issues
unloading with the non RC version.

Am I right in thinking we need to turn off coasting etc?

totalitarian, 2017-06-26:
I'll run a test in a couple of hours. 20mm cube ok?

GurliGebis, 2017-06-26:
@Sebastianv650 two questions - where does it persist the K value? (cannot find any EEPROM changes to persist it - might just be me missing it though). And didn't you forget to update the variants folder configs, to enable LIN_ADVANCE?

Sebastianv650, 2017-06-26:
@totalitarian it's based on 3.0.12 RC2, state of today. That's the MK2 branch.
Coasting should be off, that's true. Print whatever you want, if you use PLA give it a try with K50 (M900 K50). If you use other materials, you will have to do the calibration first.

@GurliGebis 
>  where does it persist the K value?

 Nowhere :) See the last half of my post above: https://github.com/prusa3d/Prusa-Firmware/issues/75#issuecomment-309014166
I don't think it's useful to have a single stored value.
Regarding the variants foder: Up to now in my branch, LIN_ADVANCE is enabled by default (see Configuration_adv.h). In my PR, it will be disabled by default. I think it's better this way until we got enough feedback to say it's OK to enable it also for i3 MK3 beginners. In the end, this decision is up to the Prusa DEV team.

GurliGebis, 2017-06-26:
@Sebastianv650 I see, didn't think about that, but you are right :) since it is on a per-filament basis, having it as part of the filament specific gcode in slic3r makes alot more sense.

Is there a reason for having it as an option that can be disabled in the firmware? (AFAIK, a K value of 0, which is the default, is equal to just the normal advance code).
Wouldn't it make more sense to just remove the old non linear advance code it replaces?

Sebastianv650, 2017-06-26:
> Is there a reason for having it as an option that can be disabled in the firmware?

I made it this way because it's the usual coding style in Marlin, how new options are handled. I also think it would be a too hard cut to make a PR that completely eleminates the known behaviour.
Basicaly, K=0 will lead to a simmilar result as a disabled LIN_ADVANCE, except some small overhead. When the Prusa team decides it's stable and can be always enabled, they can remove the ifdef statements or call me to do it.
The old ADVANCE isn't existing in Prusa FW (except some smal fragments I removed during my changes).

GurliGebis, 2017-06-26:
That makes sense :-)

bubnikv, 2017-06-26:
@Sebastianv650 I have flashed the firmware to three of our printers, we
will be testing it tomorrow. I am really curious how it will come out.
Thanks.

GurliGebis, 2017-06-26:
@bubnikv a quick off-topic question, would Prusa Research be interrested in a unified firmware, that works with both the single material and multi material version of the MK2? (Just to know if it is worth working on, or if I should just close my pull request)

totalitarian, 2017-06-26:
Flashed and a 20mm x 10mm cube. left cube was using the default prusa profile, right cube was using K50 with the settings as per [here](http://marlinfw.org/docs/features/lin_advance.html)

Sorry the filament is probably not the best to notice the differences.

One thing I did notice is the  K50 cube (right) has tighter corners, but they are also stepped a little as show in photo 2 and the edge is not square. Obviously I need to do some more tests and calibration. 
[image]
[image]

Sebastianv650, 2017-06-26:
Hard to see any difference. But at default speed that's not a surprise: The visual faces are printed at 25mm/s (external perimeters) and 30mm/s (top surface). At these speeds, LIN_ADVANCE will not do much as there is not much pressure to correct.

totalitarian, 2017-06-26:
Yeah I should have used a better colour. BTW the K50 was done at 70mm on
all visible surfaces

Sebastianv650, 2017-06-27:
I created a simple gcode which can be used to determine a good K value very quickly. No more need to print a lot of flat cubes:
- It prints a single zigzag line where each line is printed with another K value.
- First line is K=0, each following line is 10 units higher.
- Each line consists of 3 line segments in gcode: A 20mm slow segment printed at 20mm/s, a 30mm fast segment (70mm/s) and finaly a 20mm slow one again (20mm/s).
- The current gcode uses PLA temperatures and K ranging from 0 to 90. For other materials, just the tempertaures and K values has to be changed.

gcode: [K-Line Test.txt](https://github.com/prusa3d/Prusa-Firmware/files/1105704/K-Line.Test.txt) - rename to .gcode to use it.

Having a look at the final line widths, it's very easy to see what K value we need:
- If you get a thinner line width during the acceleration from 20 to 70mm/s and a slight blob during deceleration down to 20mm/s again, this **K value is too low**.
- If you get a thicker line width (blob) during the acceleration from 20 to 70mm/s and a thinner line width during deceleration down to 20mm/s again, this **K value is too high**.

Here is a labeled picture from my PLA test print. Don't get confused by all line widths on the left are thinner than on the right side - clearly a bad bed leveling. This is not related to LIN_ADVANCE ;)
- You can clearly see the signs of a too low pressure correction (too low K value) from K=0 to 20. At K=30, it's still noticeable.
- The lines with K=40 and 50 both looks good.
- The lines starting with K=60 and up have the signs of a over-correction of the nozzle pressure (K too high).
- So we can say the perfect K will be somewhere between 40 and 50. **45** will be a good value to print with.

[image]

Swiftnesses, 2017-06-27:
Excellent. Have you considered using the bed measurements as an indicator of 'k' value :p

totalitarian, 2017-06-27:
Excellent test. Here are my results, sorry they are not as obvious as your photos as I was struggling with lighting. One tip i've found is that if you rub your finger along each line you feel which ones are smoothest. I think my ideal is also between K40 & K50, would you agree? Also could I get a copy of your profile for PLA as i'm still a little unsure what needs tweaking to print with linear advance.

[image]

Sebastianv650, 2017-06-27:
@totalitarian based on your picture I would take 40. If you have a flat bed scanner (if that's the right wording), it might help with inspection in such situations when you peel of the line path from the cold bed carefuly and make a scan with highest dpi from it. Works like a magnifier!
I have no finished PLA profile, I even don't have a real "take this one" profile for my much older TAZ 5. I'm used to work with basic profiles and adjust the settings according to the object I want to print. It's more easy for me this way than to create a lot of profiles for all the possible combinations (tech. parts, figures, with and without support, ..).
That said, I there is no special setting you have to change due to LIN_ADVANCE. I would recommend to reduce the infill/perimeter overlap from 25% to 15% to increase (top) surface quality even more. Beside that, you can increase the speed using LIN_ADVANCE - but how much depends on the model kind.
Two examples:
- For technical parts (more or less straight lines with only a few arcs), you can basicaly print at the maximum flow rate your hot end can handle. I have not measured the limit of the 1.75mm e3d v6, but I guess at 0.2mm it should be save to go up to 80mm/s likely even more.
- For figures and other objects with "free style" surfaces, it highly depends on the mesh detail and the shape itseld. To get a perfect outer finish, it's important that the nozzle mooves around the perimeter lines very smoothly. Every acceleration / deceleration (even with pressure correction) will add imperfections. Therefore, it's still safe to increase speeds like infill, solid infill, support and so on up to 80mm/s. Even the perimeter lines can be printed quite fast, but the problem here is that imperfections to the inner perimeter lines will not be eleminated completely by the external perimeter, regardless how slow you print the external one. So keep the perimeter speed at a reasonable speed, maybe 50 or 60mm/s. External perimeters are the most important ones, 30-40 should be a good start point.

Sebastianv650, 2017-06-27:
Beside changing the speed, you can play with retraction settings and other things in the printer tab. For example, a wipe while retract might no longer be necessary. If your esteps/mm setting is properly calibrated, even a lift z during retract shouldn't be necessary - this was the first thing I changed in the profiles as it leads to a lot of vertical "zits" on the prints if they are not covered by the next layer. This will also reduce the print time.
I also disabled retract on layer change as I see no need for it, but your milage may vary.

totalitarian, 2017-06-27:
@Sebastianv650 Thanks. I took it from http://marlinfw.org/docs/features/lin_advance.html that there were specifics that needed to be changes. Such as setting all speeds to the same value. I guess that was just for calibration purposes?

Sebastianv650, 2017-06-27:
If you are refering to

> Set all print speeds for the cube to the same high value, 70mm/s for 0.2mm layer height for example - but be sure to not exceed the flow rate your hotend can handle!

Yes, this is only relevant for calibration. The only things you have to set or deselect permanently are described in "Before Using Linear Advance".

[78 more replies]

closed 2017-12-26 by Quintox303