prusa3d/Prusa-Firmware #85: Filament leakage during pre print Z calibration

ArnieX, 2017-05-01:
Hi, I think that printer should start heating (if not yet) before print and at certain temperature 180-190 (PLA) retract filament by ca. 4mm to stop leakage and ask user to remove anything that has leaked from nozzle. Because when Z calibration start this leaked filament get all over nozzle and sometimes I have to stop print many times before it actually start the print correctly.

Now I found out that I can retract manually which I do and it works, when printer starts print job retracted filament get back to nozzle during the flow check line.

Thanks for considering this or point me to some other solution, like maybe GCODE I can add to the start gcode section to do just this? Not sure if this is possible.

ArnieX, 2017-05-01:
Just to add, this works for me now after few tests. I have added this gcode to end script before temperature off.

G92 E0 ; reset extruder location
G1 F500 E-8 ; retract by -8
G92 E0 ; reset again

Thought it would still be great to have this value settable in printer settings.

markaswift, 2017-05-07:
Is it not easier just to heat the extruder AFTER bed levelling like I do?

ArnieX, 2017-05-07:
That might also do the trick. So to accomplish this I would just rearrange start gcode in Slic3r? Like so?

Original:
```gcode
M115 U3.0.9 ; tell printer latest fw version
M83  ; extruder relative mode
M104 S[first_layer_temperature] ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside pritn area
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
```

Changed:
```gcode
M115 U3.0.9 ; tell printer latest fw version
M83  ; extruder relative mode
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside pritn area
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
```

markaswift, 2017-05-07:
Yes, but I would suggest preheating the extruder to about 170 before doing to bed levelling - it not only makes the final target quicker, more importantly it also means any filament is softened and will not impact the levelling results.

ArnieX, 2017-05-07:
So the final GCODE should look like this?

```gcode
M115 U3.0.9 ; tell printer latest fw version
M83  ; extruder relative mode
M104 S170 ; set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S170 ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside pritn area
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
```

markaswift, 2017-05-07:
Indeed... I'm also looking to add an offset of about 100mm on the z axis before heating the bed to ensure it's not impacted by the bed temperature (right now I do this offset manually).

ArnieX, 2017-05-07:
Thanks for this idea, I will give it a try when I'll get back to my printer.

Allram, 2017-05-07:
You can also try this G-code for start script, found that it gives me no problem anymore with the mesh bed leveling, before this gcode i always had some small adjustments that i needed to do:

M115 U3.0.10 ; use the latest firmware version
G28 W ; home all axes without mesh bed leveling

G1 X10 Y-3 Z150; move extruder above bed in front for easier cleaning
M104 S160 ; set extruder temp to 160C 
M109 S160 ; wait for extruder temp
G92 E0 ; reset extrusion distance
G1 E-1.0000 F1800; Retract 1mm to prevent oozing.
G4 P30000 ; Wait for 30 secs (30000ms) for cleaning

M104 S210 ; set extruder temp to 210C 
M140 S55; set bed temp to 55C
M109 S210 ; wait for extruder temp
M190 S55; wait for bed temp

G80 ; run mesh bed leveling routine

M140 S[first_layer_bed_temperature] ; set bed temp for temperature of the first layer
M104 S[first_layer_temperature] ; set extruder temp for temperature of the first layer
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp

G1 Y-3.0 F1000.0 ; prepare to prime
G92 E0 ; reset extrusion distance
G1 X60.0 E9.5 F1000.0 ; priming
G1 X106.0 E13.5 F1000.0 ; priming

markaswift, 2017-05-07:
Why heat the extruder to 210 degrees for calibration, that would cause issues with PLA for me, and probably PETG. Basically you're using a standard bed temp for the levelling process and then heating up / waiting for cool down before printing? I like that idea.

Allram, 2017-05-07:
This is what has worked for me, and as i commented in the last post, this gives the PINDA the same preconditions as it has when you are doing the V2calibration, so the temperatures will/should not affect it. If you do mesh calibration on different temperatures, the V2 calibration will also be "off" because of the variables in the PINDA detection.

ArnieX, 2017-05-07:
I like the idea with moving and waiting for cleaning. Preheating extruder to working temp shouldn't do anything to bed levelling as only heated bed is causing some interference and is therefor good to calibrate when it is heated. Or at least that is something I read somewhere.

markaswift, 2017-05-07:
@Allram that's sound logic and makes perfect sense. I'll integrate that into my startup code! I'll likely stick to heating the extruder later and PETG is a b**** sometimes!

markaswift, 2017-05-07:
@Allram do you have experience of issuing retraction commands when the temperature of the extruder is lower than that of the filament, i.e. if I use your code and have a filament that requires a much higher heat than 210 (PETG would be borderline too). Does it cause an issue?

fiveangle, 2017-05-08:
Haven't checked the above gcode, but I've been having good luck with the following in Octoprint start gcode:

M115 U3.0.10 ; tell the printer latest firmware version available
M106 S255 ; turn on part fan full to further cool sensor to ambient if recent print just ended
M907 Z830;  resume full-power Z stepper current [my end, cancelled, and connect gcode sets to "silent"]
G28 W ; home all without mesh bed level
G1 X125 Y0 Z100 F15000; move to center front and up to keep sensor cool

-=dave

Allram, 2017-05-08:
@markaswift sorry, i have no idea of the retraction will work with a higher-temp filament. I only print with PLA at the time. But there is no problem increasing the temp in the gcode :)

ArnieX, 2017-05-08:
@Allram do you know if it is possible to set temperature with math? Like "first_layer_temp - 50"

Allram, 2017-05-08:
@ArnieX That i do not know, but that would be a killer option if that is possible :) Will do a test later today.

markaswift, 2017-05-08:
Testing RC1 firmware, how could I not see that it was already available!

closed 2018-12-11 by inquisitiveKangaroo