prusa3d/Prusa-Firmware #90: 3.0.11-RC1 temp calibration preheat doesn't happen?

tjhawkey, 2017-05-06:
Perhaps I'm confused about the instructions, but I don't see any evidence of preheat referred to when a new print is started. I'm reasonably confident that I did the calibration correctly and selected use of the calibration. I couldn't get a good z offset tune selected at all either.

mionut, 2017-05-07:
What i found from the code is that temperature calibration takes place only when printing from sd card.

markaswift, 2017-05-08:
It looks we need to reproduce the procedure via gcode when using Octoprint for example?

markaswift, 2017-05-08:
BTW, this is my current gcode:

`
M115 U3.0.11 ; use the latest firmware version
G28 W ; home all axes without mesh bed leveling

G1 X10 Y-3 Z150 ; move extruder above bed in front for easier cleaning
M104 S160 ; preheat extruder temp to 160C to soften filament
M109 S160 ; wait for extruder temp
G4 P15000 ; wait for 15 secs (15000ms) for nozzle cleaning

G28 W ; home all axes without mesh bed leveling
M140 S[first_layer_bed_temperature] ; set bed temp for first layer
M190 S[first_layer_bed_temperature] ; wait for bed temp
G4 P120000 ; wait for 120 secs (120000ms) for pinda to heat up

G80 ; run mesh bed leveling routine

G1 Y-3.0 F1000.0 ; move outside print area for priming
M104 S[first_layer_temperature] ; set extruder temp for first layer
M109 S[first_layer_temperature] ; wait for extruder temp
G1 X60.0 E9.0 F1000.0 ; prime line
G1 X100.0 E12.5 F1000.0 ; prime line
`

mionut, 2017-05-08:
You should replace G1 X10 Y-3 Z150 ;  with G1 Z150 ; 
I used this and Y axis skips steps (on prusa I3 mk2s with 3.0.11-RC1).

markaswift, 2017-05-08:
Great, have you worked out a way to implement the pinda calibration via gcode, when not printing from the SD card?

markaswift, 2017-05-08:
I actually can't see the pinda temperature calibration ever happening. I've completed the calibration, started a few prints, but the pinda is never held at the bed for 2 minutes before the levelling is started?

mionut, 2017-05-08:
I could remove card.sdrpinting == true from the line, 
`if (run == false && card.sdprinting == true && temp_cal_active == true && calibration_status() < CALIBRATION_STATUS_PINDA) {
			temp_compensation_start();` in Marlin_main.cpp
but i consider printing from sd more stable.

markaswift, 2017-05-08:
Great, I just think that most use Octoprint to initiate prints? I'm still trying to get my probe to actually do the 2 minute warm up period, it seems to just act like before I enabled temp calibration?

tjhawkey, 2017-05-08:
Wouldn't the pinda temp calibration command have to be inserted in the gcode by slic3R one way or another?

markaswift, 2017-05-08:
I don't believe the pinda temp calibration is working at all for me, not even when printing from SD, given anything I put on SD needs to come from Slic3r which contains start and end gcode, I'm not entirely clear what's at fault. Certainly it doesn't seem to work, either when printing the demo prints (batman etc).

markaswift, 2017-05-08:
Update: I ran the temp calibration again, for the first time during the start of a print it says "PINDA heating".

mionut, 2017-05-08:
Temperature calibration is a step in G80. You can change the temperature whenever you want in the startup gcode. I heat the extruder temp to 170, do home and bed leveling, then heat to first layer temperature, when the nose is down so the blob is stuck to the heatbed.

`M115 U3.0.10 ; use the latest firmware version
M83 ; extruder relative mode
M140 S[first_layer_bed_temperature] ; set bed temp
M104 S170 ; set extruder temp to 170 not to ooze and to soften the blob before homing and leveling
M109 S170 ; wait for extruder temp
G28 W ; home all without mesh bed level
G1  Z100 ; move extruder above bed in front for easier cleaning
M190 S[first_layer_bed_temperature] ; wait for bed temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
G92 E0 ; reset extrusion distance
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line`

markaswift, 2017-05-08:
@mionut why do you have the G83 and G92?

PavelSindler, 2017-05-09:
Hello and thank you for feedback. Yes, preheating PINDA probe is part of G80 (mesh bed leveling) gcode and happens when printing from sd card and temperature cal. is activated.
To reproduce this procedure when not printing from SD card, you need to move extruder to PINDA preheat position and wait for 120 s before G80. But I will improve it for usage with octoprint so you don't need to play with start gcode sequences.

@markaswift You are right that nozzle is in purge line area when preheating PINDA. When PINDA is preheated, it could happen with some materials that blob under the nozzle occurs. You don't want to have preheat position somewhere in print area, because of this blob which could probably also damage the heatbed surface little bit after long term usage. So that is why it is in purge line. 
Heating nozzle later would be an option, but it prolonges overall time.

markaswift, 2017-05-09:
@PavelSindler thank you for the upcoming tweaks and feedback, appreciated!

Can the preheat position not be moved passed the purge line (still outside of the print area)?

BTW, if we simulate it right now with Octoprint, will it still apply the pinda temp calibration after we simulate the 2 minute wait? I presumed it wouldn't as it seems to completely skip the pinda calibration routine? A revised version that we can use with Octoprint would be amazing!

mionut, 2017-05-09:
M83 was in the original Prusa Slic3r startup code, and G92 i don't know if it is necessary, i found it in another person's code. 
As @PavelSindler said, you can change position of the probe to x 70 Y -3 Z 1 and wait for 120 sec between g28 and g80. Temperature calibration values are used even if the probe is not moved to this position and a wait of 2 minutes occurs in g80(when not printing from sd), if temp calibration is active.

PavelSindler, 2017-05-09:
@markaswift Actually it will work if you add 2 minutes preheating to your start sequence. I wanted to avoid 2 minutes delay when running mesh bed leveling alone (for example from calibration menu or just by sanding gcode over serial). That is why it was done like that. But it will apply temp. caliration Z shift at the end of mesh bed leveling.
It is good to add also some retraction before heating PINDA and unretract after that (because of the blob).

markaswift, 2017-05-09:
I guess the issue is having a consistent gcode that works with both SD and Octoprint etc. When will you likely make the change (I'm available to test!).

markaswift, 2017-05-09:
guys, trying to setup the 2 minute pause... issuing G1 X70 Y-3 Z1 makes the printer move extremely slow in Octoprint, also not sure yet if it should be -3 or 0. Any ideas?

Maglin, 2017-05-09:
You don't need that long starting script.  It's over complicated and putting that in S3D causes the bed heat to turn off when ran through OctoPrint.  I just removed ` && card.sdprinting == true` from `if (run == false && card.sdprinting == true && temp_cal_active == true && calibration_status() < CALIBRATION_STATUS_PINDA) { temp_compensation_start();` in Marlin_main.cpp.  I'm not worried about some calibration taking longer as I only calibrate after a new firmware flash.  This is the starting script I'm using with now good success.

> M115 U3.0.10 ; tell the printer latest firmware varsion available
> G28 W ; home all without mesh bed level
> G0 Z150 ; move probe away from the heated bed
> M104 S[extruder0_temperature] T0 ; this will take the layer 1 temperature for extruder 0
> M140 S[bed1_temperature] ; this will take the layer 1 temperature for bed 0
> M190 S[bed1_temperature] ; wait for bed temp
> M109 S[extruder0_temperature] T0 ; wait for extruder temp
> G80 ; mesh bed leveling
> G1 Y-3.0 F1000.0 ; go outside printing area
> G1 X60.0 E15  F1000.0 ; intro line
> G1 X100.0 E12.5  F1000.0 ; intro line

markaswift, 2017-05-16:
My script works better for me... It also means I save calibrations between firmware resets etc. That and the fact it avoids oozing during calibration and pinda calibration. Each to their own ;)

PavelSindler, 2017-05-23:
This bug has been fixed in https://github.com/prusa3d/Prusa-Firmware/commit/fefa240cd7cdebd93659508aa576cf485edbf900

closed 2017-05-23 by PavelSindler

alane, 2017-05-28:
G28 W ; home all axes without mesh bed leveling
G80 ; run mesh bed leveling routine which will include 2 min pinda calibration time if it’s activated in settings

When you say "if it's activated in settings", I looked through the settings and couldn't find this anywhere.  Does the pinda calibration time automatically become set if the calibration procedure has been run, or does it need to be set somewhere in the settings in order to become active?

Thanks