prusa3d/Prusa-Firmware #94: 3.0.0.11 RC1 Strange artifacts.

grooperdude, 2017-05-15:
ever since i switched over to this beta firmware, i get these very strange artifacts when printing anything circular or cylindrical.  if it is just a cylinder by itself, the issue does not pop up. but if there is something inside of it, it happens. 
  These perfect rings of like...an extra perimeter pop up for a few layers ( all the way around the cylinder) then disappear.   i printed a calibration cube along with a cylindrical object next to it with absolutely no issues. 
 Things i have done to disagnose the issue:
1. changed values from Relative to Absolute extrusion 
2. changed filaments to make sure i diddnt get a bad spool, 
3. made sure i was not getting temperature fluctuations + cooling fan was constant. 
4. i know my zrods are perfectly straight. 
5. printed simple and complext objects, the simple object did not present the problem. 
only the complex ones. 
  The attached file is from the shell of the normal sized lion with a formable mane. 
the small lion version presented the  same issue.  then i ran the calibration version 3 hexagon that is popular on thingiverse,( with overhangs, bridge, layer test and an amphitheater looking thing on it) the cylinder on that print presented the same issues.  The calibration Cube along with the simple cylider did NOT have the issues. it only pops up when i am printing something complicated. 
 
If anyone can help me diagnose this issue, it would be Greatly Appreciated. i have not seen anything like this before or on any of the other reprap forums. 

The only thing i can think of is there is a firmware issue with the beta. 3.0.0.11 RC1 version. 
So i figured i should report it here. 
[image]

grooperdude, 2017-05-15:
Small Update... I reinstalled the 3.0.10 main firmware release that originally came with my MK2S. the problem is still happening. The GCODE preview does not show anything out of the ordinary. I have no clue why this is happening.

markaswift, 2017-05-16:
So this is unlikely related to the RC then, perhaps you can change the title to reflect this?

grooperdude, 2017-05-16:
ah yes good idea. sorry about that

mionut, 2017-05-17:
It is clear that is related to z axis movement, try to loose the acme nuts and position them in a way that they do not apply lateral force on the screws and also do not touch the z rods(do not try to center the rods on the upper bracket, it it enough not to touch the margins). Before that, move z axis all the way up, and check that the screws that apply lateral pressure to z axis, on the rods for x axis do not deform z rods(after they make contact with rods, only 1 turn or half a turn is enough).
You can also try to put some grease on the interior of the z axis linear bearings, or some machine oil on the z rods.