prusa3d/Prusa-Firmware #95: Temperature calibration has not been run yet

mariuszlabedzki, 2017-05-16:
Hi

I have info as in subject after power on printer, I made temp calibration twice without luck. Do you have idea where is the problem ? 

Thanks very much

mariuszlabedzki, 2017-05-16:
I forgot mentioned that my printer is running on 3.0.11-RC1 firmware

PavelSindler, 2017-05-17:
There was problem with storing calibration status. Have run V2 calibration or xyz calibration after temp. calibration? 
Are you printing from SD card?

mariuszlabedzki, 2017-05-17:
Hi, thanks for response, actually I ran both. You mean was issue in code or do you think this is problem on my printer ? 
xyz calibration probably delete temperature calibration results just like "z level", but v2 not reset any stored data right ?

mariuszlabedzki, 2017-05-17:
and yes I always print from sd card

PavelSindler, 2017-05-17:
IN RC1, running V2 calibration and xyz calibration leads to overwriting information that temp. calibration has been run. It is bug, which will be fixed in next release, which should be this friday.
If you would like to test temp. calibration in RC1, you must run it after other calibrations. You can of course still use live adjust Z.

mariuszlabedzki, 2017-05-17:
thank you, Its clear for me now

mariuszlabedzki, 2017-05-17:
do you want me to close this ticket ?

PavelSindler, 2017-05-17:
No, thank you. I will close it as soon as the fixed firmware version will be present on github ;-)

markaswift, 2017-05-19:
@PavelSindler is it release day? :)

PavelSindler, 2017-05-23:
Calibration status for PINDA was made independent in commit https://github.com/prusa3d/Prusa-Firmware/commit/bb3887fe1baa4d8e2a3a60e662b0106509227dd0. In https://github.com/prusa3d/Prusa-Firmware/commit/1813819c6fd27b55e97ff82d78badf75cadca7ea there was made correction for calibration status to make upgrading from previous firmware version trouble free.

closed 2017-05-23 by PavelSindler