prusa3d/Prusa-Firmware #96: M150 Command Support

slider162, 2017-05-17:
Running Octoprint LED Strip plugin, which relies on the M150 command, I get an invalid command back.  Is the M150 command supported?

http://plugins.octoprint.org/plugins/ledstripcontrol/

3d-gussner, 2017-05-18:
@slider162 The M150 gcode is in the Prusa firmware but it is disabled by default in the 'Configuration.h' file. Search for 'BLINKM' in the files, which can be found in 'BlinkM.cpp', 'BlinkM.h', 'Configuration.h' and finally in 'Marlin_main.cpp' you will see that it is gcode M150.
If you want to use it you have to de-comment '//#define BLINKM' in the 'Configuration.h' file and compile the firmware yourself.

slider162, 2017-05-18:
Nice, thanks for looking this up.  Never compiled one of these before.  Here is my plan. The general instructions could use some help.

- Install Arduino (1.6.13) from [previous releases](https://www.arduino.cc/en/Main/OldSoftwareReleases#previous).  Latest is 1.8.2 but there isn't anything for this in the Prusa zip.
- Get the rambo addon from driver page: http://www.prusa3d.com/drivers (the last link at bottom right which redirect to: http://prusa3d.com/downloads/others/ram ... _addon.zip)
- Unzip this file and put the marlinAddon folder in C:\Program Files (x86)\Arduino\hardware you should then have a C:\Program Files (x86)\Arduino\hardware\marlinAddon folder.
- Delete/rename the C:\Program Files (x86)\Arduino\libraries\LiquidCrystal folder
- Download the MK2 branch of the firmware in a zip file and unzip it (you can go to a specific commit) from https://github.com/prusa3d/Prusa-Firmware/tree/MK2
- Copy the 1_75mm_MK2-RAMBo13a-E3Dv6full.h file from the "Firmware\variants" folder of the "Firmware" folder and rename it to "Configuration_prusa.h"
- Start Arduino software
- In Tools, choose Rambo card as card type
- Load the Firmware.ino file from the unzipped folder
- Compile with voodo chicken legs :), ignore the bootloader file not found, you should have a ~180k file generated (Sketch uses 213,466 bytes (82%) of program storage space. Maximum is 258,048 bytes.
Global variables use 5,659 bytes of dynamic memory.)

3d-gussner, 2017-05-19:
@slider162 If you are using Windows try https://github.com/aderusha/Get-PrusaFwDevEnv 
Really nice powershell script getting everything you need to compile Prusa Firmware.
If you 'download' from the Prusa Github you will have the 'newest' not the last 'official' version of the firmware. So if you want to use the latest 'official/stable' version download the 'source.zip' file from the release tab in github. https://github.com/prusa3d/Prusa-Firmware/archive/v3.0.10.zip
Good luck and let us know if it is working.

mionut, 2017-05-19:
It should have ~600k, rename LiquidCrystal library from arduino(i obtained small hex size when I didn't rename it), and do not ignore bootloader not found, because it causes not to be able to program from usb next time, and you have to buy an external programmer. The bootloader is just in another folder.

closed 2018-12-11 by inquisitiveKangaroo