prusa3d/Prusa-Firmware #97: 3.0.11-RC1 - Print time is not showing when printing via USB - Octoprint

martinowren, 2017-05-19:
3.0.11-RC1 - Print time is not showing when printing via USB - Octoprint

PavelSindler, 2017-05-22:
Did you start your print using Octoprint or manually from printer?
Can you see M27 requests in your Octoprint terminal? There is example how it should look:
```
Send: N61 M27*33
Recv: GoPro_session_Lens_Cap_PLA.gcode
Recv: SD printing byte 12671/151667
Recv: 00:03
Recv: ok
```

m-ahrens, 2017-06-18:
In the forum is a post showing the situation: http://shop.prusa3d.com/forum/original-prusa-i3-mk2-f23/new-firmware-3-0-11-mk2s-mk2-t4478-s30.html#p35137

cabbey, 2017-06-18:
In the Forum post referenced above both jobs were started from octoprint, I do not see any `M27` commands being sent in the logs. Same gcode file, same octoprint install, before 3.0.11 was installed on the printer had a correctly functioning elapsed time in this scenario.  (thanks for the pointer to the issue @m-ahrens. )

Attached is the log at the start of the print in case there's anything in it that helps.
[octoprint-log.txt](https://github.com/prusa3d/Prusa-Firmware/files/1083451/octoprint-log.txt)

cabbey, 2017-06-19:
Probably expected, but just to confirm: printing from SD via Octoprint still has the elapsed time.

njcrawford, 2017-08-26:
According to http://reprap.org/wiki/G-code#M27:_Report_SD_print_status M27 is used to request SD card print status. I don't think OctoPrint should be sending M27 when it is printing vis USB, should it? That would seem counter intuitive to me, because OctoPrint already knows how many bytes have been sent over USB and how many are left.

closed 2018-12-11 by inquisitiveKangaroo