prusa3d/Prusa-Firmware #101: 3.0.11RC1 printing multi color display error

grooperdude, 2017-05-20:
I noticed when the PINDA probe is heating up right before the print, the count down starts showiung numbers in the thousands. granted, it still counts down using the correct time. it just randomly says like 1286 then back to like..90 the next second.  so far this has only happened when i use the Prusa "color print" tool. And everything still operates correctly.  
( Side note... now do i know if the LIN_ADVANCE function is working?)

GurliGebis, 2017-05-20:
I don't think asking about issues with LIN_ADVANCE in the issues list will work, until it is merged into the official firmware :)

Regarding the other issue, can you try and do a diff of the gcode to see what is changed by the tool?

PavelSindler, 2017-05-23:
Thank you for reporting this issue. I've observed similar problem after last preheat time digit. I have added printing space character there in commit https://github.com/prusa3d/Prusa-Firmware/commit/67c3e75a201754200544eb34ce7578f91d064a45 but I am not sure now if it solves this issue. I will take a closer look on this.
LIN_ADVANCE is currently not working.

closed 2019-01-29 by inquisitiveKangaroo