prusa3d/Prusa-Firmware #103: Add "Clean Nozzle" option during the print dialog

Ludio, 2017-05-22:
There should be a "Clean Nozzle" option which moves it to home up and out of the way of your print. I find that PETG is the biggest offender for a dirty nozzle and almost always by the time the first 4 layers are filled you will have some build up on the nozzle (which will drop off in the middle of a fill).

I have tried adjusting flow rate and first layer height but PETG is just sticky stuff. 

Conceptual Behavior: 
1. During a print the user selects "Clean Nozzle" from the LCD interface
2. Current position X,Y,Z are stored and heat remains at print temperature to prevent warping.
3. Nozzle is raised up (as long as their is room on the Z axis) and goes to the home 0,0 position
4. Use a Q-Tip to clean your nozzle buildup.
5. Select resume to go back printing.

This would be a huge improvement when printing to be able to prevent a glob from falling off the nozzle.

fiveangle, 2017-05-23:
There is already a "change filament" option that will allow you to do this now.  Of course with it comes the requirement to purge the filament in the nozzle, which isn't needed [shrug]

m-Th, 2017-06-29:
...besides that „Change Filament” is not discoverable for the user.

closed 2019-01-29 by inquisitiveKangaroo