prusa3d/Prusa-Firmware #108: Problems with loading/unloading/extruding filament

14422, 2017-05-22:
Just installed the "prusa3d_fw_3_0_12_rc1" firmware (13a version for my i3 MK2) and was about to start printing good old test boat and the following problems began:

1. When starting to print a first layer, filament doesn't end up on the bed but instead the slowly curls and gathers around the nozzle. Heat levels were ok from both bed and nozzle. Tested this 3 times with re-sliced and exported gcode.

2. While stopping the print the filament moving motor carved and ruined about 8cm piece of filament from one side before giving it out. This happens every time I stop printing.

3. Loading filament seems to have 5 options now, All, 1, 2, 3 and 4. The four first spin the motor the wrong way when it's supposed to drag my filament in. Only 4 works the right way.

My previous firmware version was prusa3d_fw_3_0_10 and I had no this kind of problems with it.

NikoDelarich, 2017-05-23:
I don't think 3.0.12-rc1 is supposed to be used with regular (non-MM) MK2(S) printers.
It doesn't seem to be stated on the release-page, but the release-mail said:
This firmware doesn't work with standard single material i3 MK2 printer."

PavelSindler, 2017-05-23:
Yes, It seems that you installed multimaterial firmware into standard single material printer.

14422, 2017-05-23:
Apparently that was the case. I wasn't careful enough. Thank you guys.

closed 2017-05-23 by 14422

JohnOCFII, 2017-05-25:
It seems risky to have the same firmware name for firmware that won't work with the standard printer.  Will you be able to modify the firmware to check for printer type, and then work with either device?  If not, I suggest you consider a slightly different name for the multi-material branch, otherwise you will have a support issues swamping your helpdesk in the future.

GurliGebis, 2017-05-25:
I agree, it it is possible, I would suggest going for having the same firmware for both single-material and multi-material, and then just having a menu option for switching around.
Then store a parameter in the EEPROM, and use it to decide how the code should work.