prusa3d/Prusa-Firmware #109: Printing with TEMP calibration ON fails with ABS

fns, 2017-05-24:
If I switch TEMP calibration ON (after completing the temp calibration process) and start printing with ABS the following happens:
- bed preheats
- PINDA preheats
- bed leveling starts and fails with the following error message: "Bed leveling failed. Sensor disconnected or cable broken. Waiting for reset."

I'm pretty sure neither the sensor nor its cable has any issues. If I switch of TEMP calibration everything goes well and the printer prints. If I leave TEMP calibration ON and use PLA for printing the printer works well too.

I have no idea what's going on here.

PavelSindler, 2017-05-24:
You can check if your PINDa probe is not permanently turned on due to temperature. There is red LED diode in top part of pinda probe. If it lights, PINDa is turned off, if LED doesn't light, PINDA probe is turned ON (detects target). Some PINDa probes are too sensitive and can trigger just due to temperature, without proximity of metal object. We test all PINDa probes now and don't use those with strong temperature dependency.

fns, 2017-05-24:
OK, I'll check it but what if my PINDA probe does not work with this TEMP calibration? How can I print ABS? Shall I ask the manufacturer to send a replacement probe?

fns, 2017-05-24:
So I have checked the sensor and I can confirm the LED on top (bottom) of the sensor turns off when the PINDA heating countdown reaches 20 seconds. That's a major issue for me because I cannot use the printer with ABS at all. Can you specify which PINDA sensors are compatible with the TEMP calibration procedure?

PavelSindler, 2017-05-25:
You can turn off temp. calibration and use the printer with ABS the same way as with previous firmware versions.
Temp. calibration is experimantal function which may not work with some older PINDa probes (which were not tested for temperature shift), that is why it can be turned on and off in menu. Turning temp. cal. off and on again has no infulence on temp. calibration data itself (there is no need to run temp. calibration again). So you can decide, when you want to use this feature and when not to use it.
You can also contact our support. They should send you new PINDa probe.

fns, 2017-05-25:
Just to make this clear:

- If I want to print ABS reliably (=w/o warping) I have to put the printer into a closed chamber.
- If I put a printer into a closed chamber the PINDA probe errors becomes even more drastical and I have to adjust Z offset before every print as soon as the print starts otherwise the firt layer and hence the entire print fails.

I cannot do anything else but hoping for a solution which corrects the errors of the PINDA probe. That's why I was so happy when I realised you started to make something with this issue in the 3.0.11 firmware.
On the other hand I have contacted with the support and got a promise they will send a replacement PINDA sensor. I really hope this will fix the issue.

Finally I'd be really curious how the manufacturer can print the printer's ABS parts without any warping issues.

PavelSindler, 2017-05-25:
My advice is use brim, turn off print fan, clean the heatbed surface with isopropylalkohol and if this is not enough, use aceton instead and adjust first layer heigth correctly.

fns, 2017-05-25:
Pavel, I completely understand you and thank you for the advices - I ran through on all of them with a customized Ultimaker 1 in the past. Right now my fan turns off and I'm cleaning the PEI sheet with isopropyl-alcohol before every print but I don't want to use ABS juice / acetone anymore.  

When I bought a Prusa i3 MK2 Original ™ with 
- auto mesh bed leveling
- optimized evenly heated bed 
- and PEI surface
I was expecting to print ABS parts without using brim, acetone, kapton tape, etc. I have also built a closed chamber to the printer to avoid any temperature fluctuations due to accidental air movement.

Now if I want to keep auto mesh bed leveling as one of the most attractive feature of this printer I have to manually adjust Z offset every time before I start printing from ABS because of this PINDA issue. And what is even worse, if the printer is idle long enough between prints and it cools back to room temperature I have to reset Z offset before the next print because due to the previous big offset the nozzle will scratch the printbed otherwise.

All in all, I'm a bit disappointed now but still hoping the new PINDA sensor with the experimental TEMP calibration will solve this issue.

Thanks for your support.

macfly1202, 2017-06-11:
@fns : I've exactly the same issue you describe with ABS : always need to Z live adjust during print. My mk2s have 2 months and do that since f3.0.10. The more the number of print in the same days, without powering OFF and colding my prusa i3mk2s, the more I need to play with Z live adjust (increase Z -0.6 become -0.9 become -1.2 etc at each new print for same result). You could have a perfect first layer, and the next print, few minutes after ending the first, a totally bad first layer, not "glueing" at all to heat bed.
I add your comment in this issue to because I was wondering why Z live adjust disappear from menu. 
https://github.com/prusa3d/Prusa-Firmware/issues/120#issuecomment-307654691

fns, 2017-06-14:
I got the new PINDA sensor from Prusa3D.com which now works with PINDA preheat feature. 
Unfortunately this preheat tuning still did not solve the issue: if I print several ABS parts in a row I have to re-align Z offset every time.

macfly1202, 2017-06-14:
@fns: ok it's' confirm what I say and already reported to Prusa support as what seem to be for me as a characteristics  Atmel 2560 EEPROM issue or something like that.

markaswift, 2017-06-20:
I also have this issue, such a shame as I really hoped the new temp. calibration would be a solution for drift :(

fns, 2017-06-20:
I'm so curious how the company manages the continuous printing of Prusa i3 MK2's ABS parts with this device!

GurliGebis, 2017-06-20:
My guess would be that they calibrate them once, store the values, and then don't run the calibration before each print.

fns, 2017-06-20:
This sounds very funny: you can buy a printer from them with full automatic XY and mesh level Z calibration and in order to use the same device for reliable manufacturing they are disabling continuous Z calibration?-) Even if this is the case how they prevent the offsets caused by the micro-movements of the printbed while removing printed ABS parts?

To me as an enduser this is truly a disappointment. I bought a printer to use it for reliable ABS printing and now I have two choices:
1. I have to wait for complete cooldown before printing again
2. I have to calibrate once then manually comment out G80 from start GCODE to disable further auto Z calibration

macfly1202, 2017-06-21:
I think this issue must be seriously studied by Prusa. Like @fns, I choose Prusa to have reliable print and even if it's nearly true with PLA, it's clearly not the case with ABS. Too complicated to have consecutive successful ABS print...

GurliGebis, 2017-06-21:
I haven't been printing with ABS yet, only PLA and PETG.
But from what I have seen with PETG, cooldown from 240/80 degrees only takes around 5 minutes - isn't that the case with the temperatures ABS prints at?

fns, 2017-06-21:
I print ABS with the following parameters:
hotend 1st layer: 255 ℃
hotend 2+ layers: 245 ℃
heated bed: 100 ℃
To print ABS I have to put the printer into an enclosed chamber. The chamber is not heated but completely closed, the temperature of the chamber barely reaches 42 ℃ (which is the answer to the Ultimate Question of Life, the Universe, and Everything).

In order to avoid Z offset height readjustment between prints I should wait until the hotend, bed and chamber cools back to at least 27 ℃. Trust me, I have spent weeks to find out this minimum value.
I tried almost everything to speed up the cooldown process:

- I modified the start/stop GCODE script to move the PINDA sensor away from the heated bed
- all the chamber walls as well as the top was designed to be completely removable to speed up the cooling process
- I used a 25 cm diameter fan also to speed up cooling
- I also tried to apply compressed air and icing spray on the PINDA sensor to get rid of this bloody heating data turbulence - did not help too much but now the sensor smells good ;)
- I tried the new TEMP calibration feature in the 3.0.11 firmware. My PINDA sensor was unable to handle it so I asked a replacement from the manufacturer, they have sent a new one which worked, but the TEMP calibration itself did not help at all.
- I started to design a custom mount for the PINDA sensor with a Venturi tube based cooler which might help to keep the sensor's temperature on an failsafe level - but I did not finish this part.
I'm struggling with this issue for months by now and unfortunately I cannot see any other solution than to buy/build a new printer from scratch specifically for ABS prints because Prusa i3 MK2 simply not able to serve my needs. Prusa i3 MK2 uses the PINDA sensor for XY alignment calibration as well I cannot change the sensor to a BLTouch or IR based one.

In March, 2017 I have asked @josefprusa on Twitter:
>@josefprusa Do you think the increased temperature in an enclosed chamber can affect PINDA probe’s accuracy on MK2?
The answer confirmed my worries:
>yes as with any inductive sensor

markaswift, 2017-06-22:
Hopefully @josefprusa will chime in and offer some advice.

domesticatedviking, 2017-08-23:
I am also having this issue.   Pinda temperature sensitivity is a major pain.

x4FF3, 2017-08-25:
having the same issue. i have to add always -0.6mm on every print, to have a working first layer...

optlink, 2017-10-16:
I also have the issue. I had to disable the feature in case I forget to adjust and the nozzle crashes.

an565, 2018-06-26:
I have this problem as well, and its very annoying. I have three 3D printers. The other two have simple bed leveling knobs which I calibrate with a piece of paper. They always work. The Prusa, almost never is the correct z height. Just upgraded to 3.1.0 so I can print out the parts for the MK2.5 upgrade, and suddenly now it doen't work until I turn off temp. calibration.  If MK2.5 doesn't fix all this, its going out the window. The PINDA needs to be finally dropped for a simple mechanical contact switch.

fns, 2018-06-26:
@an565: I have upgraded to MK2.5 in which the PINDA sensor got a thermistor - I hope it will solve your issues too. Before my MK2.5 upgrade I have modified the startup GCODE to 
- heat up the hotend to 170 ℃
- move the hotend to 60 mm height to avoid PINDA heatup
- heat up the bed to 100 ℃
- do the mesh bed leveling
- heat up the hotend to the final temperature
- start the print

Here it is (placeholders for Simplify3D):

```
M301 P19.09 I1.21 D75.53 ; my custom PID tuning on MK2.5 for E3D heater with silicone sock ON
G28 W ; home all without mesh bed level
G1 Y210 Z60 ; get head above bed to keep it cool
M83  ; extruder relative mode

M104 S170; heat up extruder to 170 C
M140 S[bed1_temperature]; set bed temp
M190 S[bed1_temperature]; wait for bed temp

G28 W ; home all without mesh bed leveing
G80 ; mesh bed leveling
M104 S[extruder0_temperature] T0; set extruder temp
M109 S[extruder0_temperature] T0; wait for extruder temp
G1 Y-3.0 F1000.0 ; go outside pritn area
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
```

I'm using this method with MK2.5 as well. The MK2.5 upgrade will make some difference but it will still not provide the perfect mesh bed leveling: if you print from ABS continuously and won't let the bed to cool down there still will be some minor Z height miscalculation and it may cause some warping on big objects.

My advice is to wait with the next print until the bed cools back around 30-40 ℃ and start the print.

On the other hand: go for the MK2.5 upgrade - the Z calibration seems more reliable to me and the spring steel plate alone worth the upgrade!

an565, 2018-06-26:
@fns Thanks so much. I'm finally printing the MK2.5 pieces, so hope to have it running in a week.  I guess what frustrates me most is that this "changeable Z height" has been a problem since the beginning, yet this printer is often named as the best, yet this problem is rarely if ever mentioned.  Hopefully MK2.5 will at least improve things. ABS is not the only material with a high bed temp., many others do as well. I will try your Simply3D code.  Thanks.

closed 2018-12-11 by inquisitiveKangaroo