prusa3d/Prusa-Firmware #111: XYZ calibration on 3.0.11

madhunm, 2017-05-25:
The XYZ calibration on 3.0.11 fails while 3.0.10 passes with no issues.

No physical changes made other than flashing the new firmware.

After 3.0.11 fails; uploading 3.0.10 without making changes results in a pass.

Attached are the log files of both firmware reporting XYZ calibration.

e4b_11_Cal.txt is the dump from 3.0.11 and e4b_10_Cal.txt is the dump from 3.0.10

[e4b_11_Cal.txt](https://github.com/prusa3d/Prusa-Firmware/files/1028560/e4b_11_Cal.txt)
[e4b_10_Cal.txt](https://github.com/prusa3d/Prusa-Firmware/files/1028561/e4b_10_Cal.txt)

das300hub, 2017-05-25:
Could it be that the newer calibration is seeing something the older version was not seeing correctly or calibration resolution has improved . Is it suggesting why failure occured , i cannot open the logs.

JohnOCFII, 2017-05-25:
I was able to open the two log files.

madhunm, 2017-05-25:
@das300hub There is nothing in the logs to suggest why the failure occurred. Actually, the dump isnt as verbose as 3.0.10

PavelSindler, 2017-06-08:
Thank you for logs. I am currently working on it and it is really helpful.

sebmatrix86, 2017-06-14:
Hello madhunm,

How you activate logging output during the calibration ?

I tried but I have only that : 

```
start
echo:echo: Last Updated: May 19 2017 18:53:25 | Author: (none, default config)
Compiled: May 19 2017
echo: Free Memory: 2483  PlannerBufferBytes: 1280
echo:Hardcoded Default Settings Loaded
echo:SD card ok
echo:enqueing "G28 W"
echo:enqueing "G1 Z15"
echo:enqueing "M84"
Invalid bed correction matrix. Resetting to identity.
Invalid bed correction matrix. Resetting to identity.
echo:enqueing "M45"
Iteration: 1
Iteration: 2
Invalid bed correction matrix. Resetting to identity.
```

Thanks in advance.

closed 2018-12-11 by inquisitiveKangaroo