prusa3d/Prusa-Firmware #112: [Enquiry]Linear advanced and z-offset

sarf2k4, 2017-05-26:
Hi, I would start with the z-offset, is there a g-code to adjust to change the z-offset value based on the number given?

for example my current z-offset is at 500, if I send out command for example M408 -25, the results would be z-offset at 475.

I am asking this because I opened a request at [here ](https://github.com/prusa3d/Slic3r/issues/319) since different material has different z-offset settings, even if it is same material for example abs, from different manufacturer also might have different z-offset settings. Provided that the user knows the exact z-offset settings. Using an absolute value change is not recommended because we had to change these "per flament z-offset" setting in slic3r settings should there be change in base z-offset or running a full calibration procedure again.

Next is the LInear advanced, does Prusa firmware uses linear advanced feature? If it is, would it be possible to enable eeprom variable for linear advanced K value? A bit of info also can be found in the link above.

Thank you in advance

GurliGebis, 2017-05-26:
Regarding the Linear Advance feature, that has not been implemented yet.
I have a pull request waiting, that needs to get fixed - @madhunm is taking a look at fixing it.

sarf2k4, 2017-06-05:
now that [this](https://github.com/prusa3d/Slic3r/issues/319#issuecomment-304261877) has been implemented, is there a g-code to modify the z-offset based on what I described above?

bubnikv, 2017-06-05:
G92 Zxxx should do what you want.
http://reprap.org/wiki/G-code

GurliGebis, 2017-06-05:
@bubnikv so if I have a Live Z of -0.250 now, and reset it to 0, and then add G92 Z0.250 to the generated g code at the spot after the homing, I should get the same result as if I had done nothing, correct?

GurliGebis, 2017-06-05:
Or would M206 be better for that?

sarf2k4, 2017-06-08:
May I request that reprap.org/wiki/g-code be updated to include prusa firmware as well? Since Prusa firmware uses g80 for mesh leveling/auto leveling, marlin uses g29 I suppose that this can be a good reference without the need to open up a new "issue"

Furthermore, another file or in the code description "readme.md" with list of prusa firmware's specific g-code

closed 2018-12-15 by inquisitiveKangaroo