prusa3d/Prusa-Firmware #115: Toshiba FlashAir - file protection during print, print monitoring, UI etc.

ArnieX, 2017-05-31:
Hi,

I have question to how printer handle files during print phase. On normal SD the file can't disappear mid print, but on Toshiba FlashAir user can remove the file that the printer is currently printing mid print and that could interrupt printing process. My test showed that when I remove file mid print, nothing actually happens. How is this achieved? I'm asking as if it is achieved in some way like copy the file that is currently in printing process to some hidden directory, that could well be used to detect if printer is still printing and notify user when finished.

Also as this card has quite good API, it would be possible if printer would regularly write some log file to SD card, to make Web UI that will run on the card and show current progress and some statistics data from printer. Temperature, print time, percentage, speed, etc. this would be killer feature. All this could allow for great remote monitoring without using OctoPi or similar, which I personally didn't start to like and still use safer and reliable SD print.

Hope you will like this idea and possibly make it happen. I will then do the card side of it - UI.

Thanks.

bubnikv, 2017-05-31:
> I have question to how printer handle files during print phase. On normal SD the file can't disappear mid print, but on Toshiba FlashAir user can remove the file that the printer is currently printing mid print and that could interrupt printing process.

I don't recommend doing that.

The FlashAir is formatted with a FAT file system. I suppose in most cases the file system driver on the FlashAir will not shuffle the items in the FAT table and it will not move the file blocks around, so it will likely no disturb your print, but this is not guaranteed.

> Also as this card has quite good API, it would be possible if printer would regularly write some log file to SD card, to make Web UI that will run on the card and show current progress and some statistics data from printer.

I wrote a prototype and it works. The problem is, FlashAir is discontinued by Toshiba, so it is not worth investing time into a dead product.

ArnieX, 2017-05-31:
Hi, thanks for fast reply. About that prototype, where did you get the data for it? I'm aware that it was discontinued so I'm not asking for specific Toshiba FlashAir thing, but there are plenty other WiFi SD cards, some worse, some better. So simply saving those progress data into log file could work with almost any WiFi SD card, UI itself could run on Raspberry Pi or any other server in the same network. Actually missing some kind of network connection is in my opinion biggest flaw on otherwise amazing machine.

bubnikv, 2017-05-31:
> About that prototype, where did you get the data for it?

I wrote it.

The Toshiba FlashAir has a HTTP server with CGI functionality, possibly returning values of registers, which could be written by the Marlin firmware. All the hard work is then done in JavaScript on the client side.

It would certainly make a OctoPrint like UI, but since the FlashAir is dead, it is probably worth investing time elswhere.

https://github.com/luc-github/ESP3D
https://github.com/toxuin/ESPRINTER

ArnieX, 2017-05-31:
Well I'm aware of ESP8266 monitoring firmwares, it's little too much HW hacking thought, possibility to fry expensive 3D printer electronics. Also correct me if I'm wrong but OctoPi isn't able to monitor SD card prints so I guess this is going to be the same.

bubnikv, 2017-05-31:
Octoprint works well with our printers, if you print from the OctoPrint
online, so the g-codes are stored on the SD card of the raspberry. Many of
our users use it that way.

ArnieX, 2017-05-31:
I meant that OctoPi isn't able to monitor prints that were turn on from SD card directly on printer, which is even what is recommended by Prusa Research.

closed 2018-12-15 by inquisitiveKangaroo