prusa3d/Prusa-Firmware #116: 3.0.11 Selftest fails on the Hotend Thermistor

SkyDriver2k4, 2017-05-31:
Hi,
the Selftest fails on the Thermistor. But it's working fine. Another one confirmed that he get the same error.

robrps, 2017-06-01:
I'm also see this, if I ignore it, everything prints OK, but the selftest can't continue past the failed hotend test.

PavelSindler, 2017-06-02:
Hello, thank you for reporting this. I will take a look at it.

PavelSindler, 2017-06-02:
Can you please provide the message which is printed on LCD when selftest fails?

SkyDriver2k4, 2017-06-02:
sure.
Selftest error !
Please check:
Heater/Thermistor
Not connected
Selftest failed.

PavelSindler, 2017-06-02:
And can you check nozzle temperature displayed on LCD before and after failed selftest?

SkyDriver2k4, 2017-06-02:
33°C Start
After the test 52-53°C

robrps, 2017-06-02:
Says the same for mine - video of it happening if that helps at all?
https://www.dropbox.com/s/4kq0a62lbmrmipg/2017-06-02%2017.10.17.mp4?dl=0

HenrikJay, 2017-06-02:
Same issue (MK2S).
After upgrade I got bed heating error when trying to print.
Downgraded firmware which allowed me to print but bed heat sensor is not working correctly. Takes 10+ minutes to heat up and shows 50 degrees when the bed actually is at 80. When printing the bed goes up to 108 (stopped print at that time).
Trying to upgrade again.

PavelSindler, 2017-06-06:
Thank you guys for great feedback. This issue was caused by limits which were set to strictly for quite limited time when nozzle was heated. I tested this on many printers here, and all passed, but in some cases it was very close to fail and if I compare it with @SkyDriver2k4 temperatures after calibration I can see, that this was the cause. So I just prolonged selftest heating time for nozzle and also for heatbed little bit. This will be fixed soon in 3.0.12 release. I will update info here as soon as commit will be merged here.

robrps, 2017-06-26:
3.0.12-RC2 has fixed the issue for me - thanks.

ManuGithubSteam, 2017-08-04:
@SkyDriver2k4  Please close the issue and update to 3.0.12

closed 2017-08-04 by SkyDriver2k4

d777, 2018-01-24:
I am getting this issue with 3.1.0

Xsiliconkid, 2018-01-26:
Where do i find the updated Firmware version 3.012-RC2?

jl1990, 2018-02-06:
It happened to me on 3.1.1

mptrs, 2019-05-22:
Looks like the issue is back in V3.7.1. After downgrading to V3.7.0 there was no error and everything is working fine.

chriskpalmer, 2019-06-04:
One of my twelve i3 Mk3 machines had this issue with V3.7.1 and following mptrs I rolled back to V3.7.0 and everything works fine. I found this after replacing the hotend thermistor :(

DominiqueFuchs, 2019-06-06:
Can confirm, this has to be reopened/fixed again. Ran the calibration wizard with 3.7.1 (after running the printer for a long time, this was for specific reasons uncorrelated to heating problems) and self test fails with "Heater/Thermistor not connected". Rolled back to 3.7.0 and wizard runs fine.