prusa3d/Prusa-Firmware #118: 3.0.11 Weird hotend temperature variation

Cifik, 2017-06-06:
Hello,
in firmware 3.0.11 there seems to be weird hotend temp variation which I couldn't remember from 3.0.10.
In 3.0.11 when temp is set to 225*C the real temperature cycles between 211 and 218. Never hits 225*C. The variation is also quite big I think, there are "lines" clearly visible on flat surface printed (rectilinear - 100%). The smoothness raise and fall every 7mm (guessed) - the "lines" are exactly colliding with the temperature variaton (I'm using Prusa's PLA).

GurliGebis, 2017-06-06:
Have you tried doing a PID tuning of the hotend?

Cifik, 2017-06-06:
Yes, I did, but there is no mention how it should be set or how often is should be done etc. ... I think if this feature was added to FW, it would be worth at least mentioning it somewhere.

Do you please know how it should be set? (I used the pre-set value 210 the printer offers)

GurliGebis, 2017-06-06:
I haven't done it using the firmware option, so I cannot tell if that works well or not.
What I did was follow Matt's guide here: https://mattshub.com/2017/04/26/pid-tuning/

PavelSindler, 2017-06-07:
When running PID tuning from menu, you just set temperature, confirm with pressing the knob and that is all. You just wait for calibration finish.
How to set temperature for PID calibration? Usually it is sufficient to run PID calibration for just one temperature (for example default 210 C if you use it often or somewhere in the middle of usually used temperature range) and the temperature remains stable also for other temperatures. If you print mostly ABS, you can run PID calibration for 255 C for example. Or if you want to be precise, you can rerun PID calibration always when you change metarial (nozzle temperature). It is also possible to use PID tuning gcodes in the starting sequence of your gcode file. Some users do that for the highest precision. But these are extreme cases. 
My experience is that if you have printer, where the PID calibration was never run, nozzle temperature usually cycles few degrees around target temperature for all materials (nozzle temperatures). If you do PID calibration on such printer for just one temperature (for example somewhere in the middle of used range), temperature is stable there and there is improvement for temperature stability for all materials (no fluctuation can be observed on LCD).
For example I have run temp. calibration for 210 C and I can't see any fluctuations for 210 C nor 255 C. It just remains stable in whole range. If I load hardcoded default settings back, temperature starts to change from 252 C to 259 C.
@Cifik, I would recommend to run PID calibration again. It really seems that something went wrong during calibration process. For example that it was not finished.
If it will not help, you can try to load default settings (which were used in previous firmware version before  PID calibration was available) by sending M502 to the printer and see if it helps.

Cifik, 2017-06-07:
Hello,
today in the morning I've run the PID tuning again (and again from printer menu). This time I simply waited a long time (approx 30mins - I had other duties to do) and then turned the printer off and on again.
After this the temperature seems to be stable and exactly where it should be, so my problems probably came fom some error during the last calibraion. From my point of view the problem is, that there is no mention about PID tuning in manual, neither elsewhere. And there is NO status message at all on the printer - you just set the temp and approve by button, and the printer will return to the main status screen (even hotend set temp shows 0), no status message, no info about progress or completion.
So yes, it seems to work probably, maybe I just started printing too early last time or something like that. But some message that "printer is doing something, keep your hands away" would be really nice.

closed 2017-06-08 by PavelSindler