prusa3d/Prusa-Firmware #120: 'Z live adjust' disappear from menu during print

macfly1202, 2017-06-11:
Hi, I've got a MK2S since april 2017. With firmware 3.0.10, 3.0.11RC1, 3.0.10 I encounter this issue. On long print, more than 60 minutes, after have made some change during the printing period (maybe 3-8 change) , Z live adjust totally disappear from menu. Don't know how to reproduce this but this occur quite offten. Perhaps link to temperature. Very often it happen during ABS printing. 
here a short video that show this :  https://youtu.be/tkzN5zpahEY

GurliGebis, 2017-06-11:
That is by design - the firmware checks if the current Z height is above or equal to 2.0, and if that is the case, the "Live adjust Z" menu item is hidden.
Why this is done, I don't know, but it is by design.

GurliGebis, 2017-06-11:
If you know C++, have a look at the code here: https://github.com/prusa3d/Prusa-Firmware/blob/d44b30bbc63d823d7bedfc4672ec70ab30e437d8/Firmware/ultralcd.cpp#L3513

It is configured in the configuration file here: https://github.com/prusa3d/Prusa-Firmware/blob/5d2e7fe97b629f66cd743c8dae76d950a60bd3b3/Firmware/Configuration.h#L293

macfly1202, 2017-06-11:
thanks, GurliGebis. As I see in conf file, #define Z_HEIGHT_HIDE_LIVE_ADJUST_MENU 2.0f
What could be the reason to hide this? Often it's on ABS print after 1-2 hour that I need to make some adjustments. I could compile it on my own now I know this but I would prefer to have an explanation. Thanks again

JohnOCFII, 2017-06-11:
> What could be the reason to hide this?

I'm just a user, but according to the Prusa documentation, the purpose of Live-Z is to set your first layer height to ensure good adhesion to the bed.  If that is the reason for the feature, that could explain why it is disabled once a print is assumed to be more than a couple of mm away from the bed.  

It sounds like you would like a more generic "baby-stepping" feature that could be used anytime throughout a print.

macfly1202, 2017-06-11:
@JohnOCFII : ok it makes sense for first layer. I suppose It's why dev' has decided to implement temperature Calibration.

I want to my main issue (workaround using Z livre adjust) is exactly describe by 'fns' user at https://github.com/prusa3d/Prusa-Firmware/issues/109
he said : 
"Now if I want to keep auto mesh bed leveling as one of the most attractive feature of this printer I have to manually adjust Z offset every time before I start printing from ABS because of this PINDA issue. And what is even worse, if the printer is idle long enough between prints and it cools back to room temperature I have to reset Z offset before the next print because due to the previous big offset the nozzle will scratch the printbed otherwise." 
This is exactly my issue and why I always need to play with Z live adjust.

fns, 2017-06-13:
@macfly1202 I'm re-setting Z offset only at the beginning of the prints.

macfly1202, 2017-06-27:
Ok I think you can close this issue.

closed 2017-06-27 by macfly1202