prusa3d/Prusa-Firmware #121: Blobs on Bed when doing Mesh-Bed Leveling

tbedore, 2017-06-15:
Consider in the mesh bed leveling to retract the filament out a little bit, to prevent little blobs of filament hitting the bed and sticking on all 9 calibration points, just retract 15mm at the start and -15mm after all 9 points are done, just a small little thing that would make printing very easy

GurliGebis, 2017-06-16:
I am seeing the same on my prints, so you are not alone.
I don't know if the firmware should do this, or if the slicer should do it before the G code to trigger the leveling.

michalzielanski, 2017-06-20:
Try this start g-code: https://github.com/prusa3d/Prusa-Firmware/issues/90#issuecomment-299908302

GurliGebis, 2017-06-20:
I tried something similar like that today, and it seems to have done the trick.

The default start g-code in slic3r should be updated to do it this way, to workaround this problem.

m-Th, 2017-07-05:
hMMMM.... 

> Try this start g-code: #90 (comment)

It makes some problems with some filaments. Sometimes works ok but sometimes not.

Specifically, using FormFutura GlowFill's filament, the 2nd time when the g-code from above is used, the filament is not extruded anymore. It needs Stop Job, doing a manual extrusion via Settings > Move Axis > Extruder and after this it needs to use the "normal" start g-code from Slic3r in order to be ok.

I suspect that 

>M104 S170 ; set extruder temp to 170 not to ooze and to soften the blob before homing and leveling
>M109 S170 ; wait for extruder temp

are to blame...

GurliGebis, 2017-07-05:
Why is it a problem to pre-heat it to 170? (just curious)

m-Th, 2017-07-05:
@bubnikv caught the problem in the parallel theread from Slic3r's repo. 

TL;Dr: 170 is filament specific. And more: composition specific. For me 170 it works with PLA but it doesn't work with FF's GlowFill which is in fact the same PLA but filled with phosphorus pellets. (disclaimer: FormFutura produces very high quality filaments at least IMHO).

bubnikv, 2017-07-05:
The current Slic3r PE supports a filament specific start G-code section,
you may try to utilize that.

m-Th, 2017-07-05:
Yes – indeed we can utilize that. But for this please, please, pleaaaase implement this one:

 

https://github.com/prusa3d/Slic3r/issues/377

 

Inserting the appropriate G-Code for each PLA variant (PLA+, GlowFill, BronzeFill, CopperFill, etcFill) is definitely the correct way to go, but this would increase the list of filament profiles.

 

Nothing wrong with this per se, but having such a limited width for the drop-down widget makes suddenly the things quite hard in this regard.

 

I know that it is easy to implement it and it will help a lot in UX.

bubnikv, 2017-07-05:
> K1NRgaJpZM4N7YFM> .

> I know that it is easy to implement it and it will help a lot in UX.

If it is easy to implement, then please provide a pull request.

Sorry, but a day has 24 hours only and there are currently 156 open issues,
the issues influencing the print quality having a higher priority IMHO.

closed 2018-12-17 by inquisitiveKangaroo