prusa3d/Prusa-Firmware #130: New information within support menu

markaswift, 2017-06-23:
Hi @PavelSindler,

Could you please explain the new details within the support menu. It seems we have 2 extra pieces of information:

1) Y distance from min (Mine shows 1.37, 1.48, 1.63)
2) Angle difference (Mine shows 0.00)

Thank you!

madhunm, 2017-06-23:
@markaswift are you on 3.0.12RC2? if yes, then it is showing you the position of the front row of the PINDA points and the skew as calculated by the firmware.

Congrats! you seem to have a square printer.

@PavelSindler followup question. is this angle difference counter clockwise or clockwise?

markaswift, 2017-06-23:
Hi @madhunm, that's what I guessed... Seems I can tweak it a little more ;)

I'd also like to know the direction of rotation @PavelSindler

PavelSindler, 2017-06-23:
Hello,
you understand it correctly.
When searching for position of calibration points during xyz calibration, PINDA probe needs to move freely in X and Y axis in area around calibration point. Y coordinate of three front calibration points can be difficult to measure accurately because these points are very close to minimum Y coordinate which can be reached by PINDA probe (it is limited by endstop hit).
First parameter (Y distance from min) is distance between measured Y coordinate of heatbed calibration point and Y coordinate where calibration point is considered out of PINDA probe reach and xyz calibration may be compromised.
Positive number means that front points can be reached by PINDA probe and thus xyz calibration results are accurate. 
Negative number means that front point Y coordinate was not measured accurately because point was out of PINDA probe reach. 
If "Y distance from min" shows for example:
Left: -0.1 mm
Center: 0.5 mm
Right: 1.2 mm
XYZ calibration finishes with "XYZ calibration compromised. Left front calibration point not reachable." message.
Second screen shows parameter which determines if x and y axis are perpendicular to each other or  skewed (and how severly). So all numbers between 0 and 0.12 means that you build it very well and your xyz calibration finished with "X/Y axes are perpendicular. Congratulations!" message. Number from 0.12 to 0.25 means slightly skewed axes. Everything over 0.25 is severe skew. But also this skew can be compensated by xyz calibration.
So I would say, @markaswift , that your printer is very well built and should work greatly. Congrats! :-)
Measured skew (angle difference) is absolute value. But if you have severly skewed printer and want to improve it, you can use info from first screen (distance from min.) to make your axes perpendicular.

madhunm, 2017-06-23:
@PavelSindler is there a possibility to have the angle difference be a signed value? that way, for example, you could infer that a negative value means that the bed is rotated left looking from top and a positive value means that the bed is rotated to the right.

PavelSindler, 2017-06-23:
Yes, it would be possible to make it signed value. Thank you for feedback. Our main motivation to implemented this was to make possible to debug if the printer is close to some borderline. If you have for example printer with distances from min. close to zero and you rerun xyz cal., you can easily obtain very different result from previous xyz cal. run and that can be confusing for users and also it is more difficult for us to determin if there is some possible issue in xyz calibration.

madhunm, 2017-06-23:
@PavelSindler I understand what you are saying...

However, most users do not run the XYZ calibration unless they rebuild the printer. and, when you see the direction to rotate teh bed to get to square, it would help a lot.

fiveangle, 2017-06-24:
> 1. Y distance from min (Mine shows 1.37, 1.48, 1.63)
> 2. Angle difference (Mine shows 0.00)

@PavelSindler - From the quote above, can we infer that this users front threaded rod is equidistant from the metal frame, while the rear threaded rod is slightly skewed, with X=0 side further from the metal frame than the X=250 side ?

closed 2018-12-11 by inquisitiveKangaroo